Opentrons/opentrons #234: Container not showing up for calibration

Suzyhuang, 2017-05-15:
If the only function applied to a container in the entire protocol is move_to, it will not show up for calibrations in the app.

mcous, 2018-05-17:
Ping @btmorr and @Laura-Danielle: this bug is still present in `v3` of the API. The following protocol results in no pipettes and no labware being reported to the app:

```py
from opentrons import labware, instruments

trough = labware.load('trough-12row', '3', 'trough')
plate = labware.load('96-PCR-flat', '1', 'plate')
p200rack = labware.load('tiprack-200ul', '2', 'tiprack')

p200 = instruments.P300_Single(mount="left", tip_racks=[p200rack])

p200.move_to(p200rack)
p200.move_to(trough)
p200.move_to(plate)
```

btmorr, 2018-06-21:
@umbhau this is actually a purposeful design, but I personally think it should be reconsidered. The problem with this is that changing this behavior would break some existing protocols that have been written to leverage the existing design. I am in favor of making it such that any labware that is loaded is available for calibration, but this is something that the product team should weigh in on.

umbhau, 2018-06-21:
Hey @btmorr, I'm in favor of making all labware available as well, but would like to run through the original intended benefits of the current design (definitely before my time...sure I'm missing something). 

Should also consider any unintended side effects with stacked placeables, i.e. modules.

arachnologist, 2018-08-17:
personally I'm hoping they reconsider it because my lab can't run any protocols because they all result in "running out of tips" since the app won't load any tipracks we don't call directly. This would probably be fine though if they would just fix the bug that makes calling a specific tiprack erase that tiprack's calibration data.

mcous, 2018-12-10:
@arachnologist sorry but I completely missed you comment in this thread. Can you please explain what you mean by:

> the bug that makes calling a specific tiprack erase that tiprack's calibration data

We're unaware of any such bug, so if you've observed something weird I'd love to get it ticketed so we can address it

closed 2020-10-21 by asogluizzo