Opentrons/opentrons #262: Containers disappear from step list

SimplyAhmazing, 2017-07-18:
Both Nusrat & I experienced containers disappearing at random from the calibration step list when one is using the app. The disappearing sometimes takes place after running a protocol and interacting with the 'move to' buttons on a container.

I have not found a reliable way to replicate this bug.

astaff, 2017-07-19:
Do you think that containers left could be the ones accessed by the pipette during the protocol run or move to? If so, it could be us re-populating the instrument/container map to those affected by the run/move and reloading it

SimplyAhmazing, 2017-07-20:
The containers that we've seen disappear include a trough, a container central to the protocol always used. Ignoring it until we can reproduce.

SimplyAhmazing, 2017-07-20:
One easy fix that avoids caching and cache invalidation issues that @andySigler suggested is to re-run the protocol against the null driver.

closed 2017-07-20 by SimplyAhmazing