Opentrons/opentrons #272: App fails to start on windows due to winston module not being located

SimplyAhmazing, 2017-07-20:
On windows, App fails during installation due to winston logger not finding module `./transports/file`. This seems like a packaging issue. 

[image]

astaff, 2017-07-21:
app.asar seems to contain everything needed to run the app. Can you confirm this behavior by trying to run make build and running unpackaged app on Windows? If it works, try re-packaging it with NSIS.

SimplyAhmazing, 2017-07-21:
I think we can confirm the unpackaged app works on windows because of partial running of integration tests but I can double check.

SimplyAhmazing, 2017-07-21:
Packaged app doesn't start on windows and results in the error below. This seems to be due to none of the app deps being packaged properly.

```(venv) PS C:\Users\nick\github\opentrons\app> & '.\dist\win-unpacked\Opentrons 2.0.exe'
(venv) PS C:\Users\nick\github\opentrons\app>
info: Starting App
User Data Path C:\Users\nick\AppData\Roaming\OT-App
Serving UI from:  C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar.unpacked\ui
undefined
undefined
info: Uncaught Exception:  Error: ENOENT, bin\opentrons-api-server not found in C:\Users\nick\github\opentrons\app\dist\
win-unpacked\resources\app.asar
    at notFoundError (ELECTRON_ASAR.js:115:19)
    at Object.module.(anonymous function) [as execFile] (ELECTRON_ASAR.js:215:16)
    at ServerManager.execFile (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle
.js:16136:41)
    at ServerManager.start (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js
:16124:12)
    at _callee3$ (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:15867:27)
    at tryCatch (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:9570:40)
    at Generator.invoke [as _invoke] (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist
\bundle.js:9808:22)
    at Generator.prototype.(anonymous function) [as next] (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resource
s\app.asar\shell\dist\bundle.js:9622:21)
    at step (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:15711:191)
    at C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:15711:437
    at App.<anonymous> (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:157
11:99)
    at App.startUp (C:\Users\nick\github\opentrons\app\dist\win-unpacked\resources\app.asar\shell\dist\bundle.js:15906:1
8)
    at emitTwo (events.js:111:20)
    at App.emit (events.js:191:7)
(node:9600) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read prop
erty 'spawnfile' of undefined
(node:9600) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are
not handled will terminate the Node.js process with a non-zero exit code.```

SimplyAhmazing, 2017-07-21:
I think this is because packaging app deps as `files: []` is not consistent to all OSs if I remember correctly...

SimplyAhmazing, 2017-07-21:
Although squirrel installer throws an error while installing the app, the app does get installed and works fine after installation.

Tried nsis installer and it also works fine. Not sure what auto-updating means here...

astaff, 2017-07-21:
Moved log init to onready event and it fixed the issue

commit d9504f00 (2017-07-21): Fixes #272, adds script to rename artifacts in windows build before upload
commit 8cb761ab (2017-09-21): Fixes #272, adds script to rename artifacts in windows build before upload
closed 2017-07-21 by SimplyAhmazing