Opentrons/opentrons #278: Pick-Up-Tip within UI collides with containers after 2nd click

andySigler, 2017-07-26:
Seems like the calibrated position is not being considered during a robot `arc` movement only some of the time when being called from the flask server.

Steps to reproduce:

1. Load protocol with `trough-12row` and `tiprack-200uL`
2. Calibrate `trough-12row` so that it's is higher up on the Z axis than the `tiprack-200uL`
3. Press `Move-To` button on the `trough-12row` (should well well, moving up to safe height)
4. Press `Pick-Up-Tip` button on the `tiprack-200uL` (should well well, moving up to safe height)
5. Press `Move-To` button on the `trough-12row` (should well well, moving up to safe height)
6. Press `Pick-Up-Tip` button on the `tiprack-200uL`, and the pipette should move DOWNWARDS, colliding with the trough

SimplyAhmazing, 2017-07-26:
Setup above instructions and could not reproduce this bug. Validated setup with @andySigler and the bug did not re-occur on latest release for 2.5.0. Closing issue..

andySigler, 2017-08-01:
Happened again with Lucia with latest Dev build on Windows, commit #fecbde33

astaff, 2017-09-07:
Fixed in 2.5.2, @andySigler ?

mcous, 2017-12-20:
Assuming fixed and closing the issue. @andySigler please re-open if this is not the case

closed 2017-12-20 by mcous