Opentrons/opentrons #285: `make docs` failing in Travis

IanLondon, 2017-08-07:
Failing doctests? in dev builds

https://travis-ci.org/OpenTrons/opentrons/jobs/261177716#L1119

```
tube-rack-.75ml

-------------

/home/travis/build/OpenTrons/opentrons/api/docs/source/containers.rst:374: ERROR: Document or section may not begin with a transition.

/home/travis/build/OpenTrons/opentrons/api/docs/source/examples.rst:222: ERROR: Document may not end with a transition.

/home/travis/build/OpenTrons/opentrons/api/docs/source/modules.rst:7: ERROR: Document or section may not begin with a transition.

make[1]: *** [doctest] Error 1

make: *** [docs] Error 2

The command "if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then (cd api && make docs > /dev/null && cd ..) fi;

" exited with 2.
```

astaff, 2017-09-07:
@Kadee80 @IanLondon can you reference a commit that fixes that here, and/or reference this issue in PR?

Kadee80, 2017-09-07:
We investigated today. Docs build but the template update was never merged in. I closed the old pr (from June) will make a pr in the am with updated doc templates. Will branch off master as we discussed.

Kadee80, 2017-09-07:
Accidentally closed. Will keep open till resolved