Opentrons/opentrons #288: Calibration data shifts or is erased

SimplyAhmazing, 2017-08-08:
User reported the following error:

> I've been testing the new robot today and keep running into issues with a-axis calibrations shifting and being erased. 
> The problem occurs when doing a lot of recalibrating (such as after fine tuning pipette position adjustment, then recalibrating positions for containers). 
In once instance, the a-calibration shifted down a few mm, and in another instance the calibration wiped clean. 
> Any advice on solutions for this? I am using v2.4.2 with the OT-One plus. I'll follow up this message with our torture protocol that I was calibrating for.

jerrywan4, 2017-08-31:
@astaff 

Customer sees an error in running their protocol. Their calibration seems correct in the calibration screen when checking by moving to locations, but when running, robot goes to wrong location and error is thrown.

```
5:50:18 PM - Error in: Traceback (most recent call last): File "opentrons\server\helpers.py", line 32, in run_protocol File "<string>", line 27, in <module> File "opentrons\instruments\pipette.py", line 853, in drop_tip File "opentrons\instruments\pipette.py", line 257, in move_to File "opentrons\util\trace.py", line 9, in decorated File "opentrons\robot\robot.py", line 476, in move_to File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 345, in move_head File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 339, in move File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 355, in consume_move_commands File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 384, in wait_for_arrival File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 500, in get_position File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 201, in send_command File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 219, in write_to_serial File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 246, in wait_for_response File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 280, in readline_from_serial File "opentrons\drivers\smoothie_drivers\v1_2_0\driver.py", line 298, in detect_limit_hit RuntimeWarning: Robot Error: limit switch hit
```

[Calibration bug.zip](https://github.com/OpenTrons/opentrons/files/1268582/Calibration.bug.zip)

IanLondon, 2017-09-01:
Related #306

astaff, 2017-12-20:
No. Closing.

closed 2017-12-20 by astaff

kolaworld, 2018-04-19:
Hi All,

I see that this was closed. I'll go ahead and describe my issue here but please let me know if you'd prefer I open a new one.

We are still using the OT-One S as we wait for the OT-2 to ship out. I received the above bug (RuntimeWarning: Robot Error: limit switch hit) for the first time today on the B-axis. I shut-down and restarted the robot which made that bug go away, but then the B-axis stopped working as expected. The B-axis motor sequentially goes down from the homing position instead of doing the 'Top-Bottom-Top' motion in order to aspirate.

Please let me know if there is any way to troubleshoot this.