Opentrons/opentrons #301: Instrument calibration check mark sometimes does not show up

SimplyAhmazing, 2017-08-18:
When you calibrate an instruments positions (top, bottom, blowout, droptip), sometimes the app does not show calibration check mark.

Kadee80, 2017-08-18:
I remember seeing this from time to time when we check for isCalibrated. If I remember correctly the check happens before state is updated.

SimplyAhmazing, 2017-08-18:
Huh, @Kadee80 are you saying it's something that's been around before 2.5.0 ?

Kadee80, 2017-08-18:
I saw it on our dynamic deck map branch. Which was never merged in. I assumed it was because we changed the calibration logic due to the deck.

SimplyAhmazing, 2017-08-18:
Just cleared out my calibrations data directory, opened app 2.5.2, uploaded a protocol, calibrated all positions, and was able to get the green check mark and run my protocl.

Can't reproduce this issue locally.

closed 2017-09-07 by astaff