Opentrons/opentrons #306: max volume of 1000 is set to 20?

IanLondon, 2017-08-23:
Mysterious!

```python
from opentrons import containers, instruments

p1000rack = containers.load('tiprack-1000ul', 'A1')
trash = containers.load('point', 'B2')

p1000 = instruments.Pipette(
    axis="b",
    max_volume=1000,
    min_volume=100,
    trash_container=trash,
    tip_racks=[p1000rack]
)

print(p1000.max_volume)  # prints: 20
```

(Note to self: found this because it causes `serial_dilution_htc.py` in Protocols to fail: p1000 raises `RuntimeWarning: Pipette with max volume of 20 cannot hold volume 300.0`)

lmtgalhardo, 2017-08-23:
this has been happening lately: old calibration data is messing up the pipettes. the fix I know of is to name your pipette something and then it should work. I think we should look into this though, the app used to not do this and its weird for it to be required to name your pipette

IanLondon, 2017-08-23:
It turned out this is an issue with persisted calibration data somehow having precedence over my `max_volume` argument. I had to delete the calibration folder.

btmorr, 2017-12-20:
This is currently not a problem with old calibration data. In 3.0, the `max_volume` parameter of Pipette is ignored, and `Pipette.max_volume` is calculated from a combination of `PLUNGER_POSITIONS['top']`, `PLUNGER_POSITIONS['bottom']` and `Pipette.ul_per_mm` (all in opentrons.instruments.pipette). The main problem for using non-default pipettes appears to be the fact that PLUNGER_POSITIONS is hard-coded. I believe work is in progress to include plunger position data, ul_per_mm and any other parameters like this in data that can be read from the pipettes themselves, which will resolve both of these issues.

closed 2017-12-20 by btmorr