Opentrons/opentrons #307: Not aspirating/dispensing after a move_to() command using direct strategy

liulisa, 2017-08-23:
We tried to modify the [precision pipetting](http://docs.opentrons.com/examples.html#precision-pipetting) example to aspirate instead of dispense. However, it will move as needed, but it won't actually aspirate anything if we use the 'direct' strategy. However, if we let it move using the default ('arc') strategy, it doesn aspirate.

theosanderson, 2017-10-20:
I also have this problem. Adding a delay did not help - any solutions, even hacky would be appreciated.

mcous, 2019-10-09:
Categorizing as a bug out of an overabundance of caution. I have no idea if this is still an issue or not; we should test it out

SyntaxColoring, 2020-05-02:
I assume the aforementioned "precision pipetting" example is [this one](https://docs.opentrons.com/v1/examples.html?highlight=precision#precision-pipetting).

theosanderson, 2020-05-02:
> I assume the aforementioned "precision pipetting" example is [this one](https://docs.opentrons.com/v1/examples.html?highlight=precision#precision-pipetting).

Yes that didn't work (haven't tried to replicate on OT2)

asogluizzo, 2020-11-20:
Close after 4.0.0 release

closed 2021-01-26 by mcous