Opentrons/opentrons #311: build ot app on clean ubuntu 17.04

cfblaeb, 2017-08-28:
Hi,
I had quite a few issues trying to build ot app on a fresh install of ubuntu 17.04.
I thought I would share the required steps so you could update your build instructions.
Mind you I have tested this on the actual robot yet, so there may be even more issues.

In addition to install python (via pyenv) and node (via nvm) I also had to install the following:
libgconf-2-4, graphicsmagick, icnsutils

The python I installed via pyenv had to be compiled with "enabled-shared":
env PYTHON_CONFIGURE_OPTS="--enable-shared" pyenv install 3.6.2

I couldnt get it to work with the recommended npm 6, but it worked with the latest npm 8.

With npm 8 it didnt automatically install dependencies in the app/ folder so I had to run "npm i"

Now I'm gonna go try and plug the robot in =)

astaff, 2017-09-07:
Hi @cfblaeb ! Thanks for your feedback. Added this to our `develop` branch. Let us know how things work with the robot plugged in.

closed 2017-09-07 by astaff

cfblaeb, 2017-09-08:
It works. Although, I am having issues with the magdeck but that may be unrelated to this.