Opentrons/opentrons #314: create() function generate json wells out of order

lmtgalhardo, 2017-08-28:
This code's create function:

`from opentrons import robot, containers, instruments

# only change the values within these boundaries

########################start#########################

# This opens the csv file with absolute path (will be different depending on operating system)
# file paths on Windows look more like 'C:\\path\\to\\your\\csv_file.csv'
# on Mac more like '/path/to/your/csv_file.csv'
path = 'C:\\Users\\calel_000\\Libraries\\Documents\\Opentrons\\Protocols\\exampleGBF.csv'

# number of tubes of sample (not counting empty pooling tubes)
sampletubes = 9  # can change this number here

# number of resulting tubes you are harvesting (the empty tubes that get 3 samples in each)
harvest = 3  # can change this number

# number of wells to dispense DILUENT DISTRIBUTION COMMAND on HCV ("13") plate
diluentnum = 16  # can change here 

# number of wells to dispense conjugate for each plate
HCVconjwells = 16  # can change here
HIVconjwells = 4  # can change here
HCGconjwells = 4  # can change here
HBsAgconjwells = 8  # can change here

# number of wells to dispense substrates for each plate
HCVsubswells = 16  # can change here
HIVsubswells = 4  # can change here
HCGsubswells = 4  # can change here
HBsAgsubswells = 8  # can change here

# number of wells to dispense during stop command each plate
HCVstopwells = 16  # can change here
HIVstopwells = 4  # can change here
HCGstopwells = 4  # can change here
HBsAgstopwells = 8  # can change here

########################end#########################

# number of samples combined in one tube
poolnum = 3

# volumes for varieties of elisa 
HBsAg_vol = 50 
HCV_vol = 10 
HCG_vol = 100 
HIV_vol = 100

# volumes for conjugates and substrate
conjsub_vol = 50
        
# length of sample rack row
rackrow = 4

containers.create(
    '4x6_tuberackGBF',                    # name of you container
    grid=(4, 6),                    # specify amount of (columns, rows)
    spacing=(6, 10),               # distances (mm) between each (column, row)
    diameter=12,                     # diameter (mm) of each well on the plate
    depth=75)                       # depth (mm) of each well on the plate

containers.create(
    '3x4_vialrackGBF',                    # name of you container
    grid=(3, 4),                    # specify amount of (columns, rows)
    spacing=(10, 15),               # distances (mm) between each (column, row)
    diameter=20,                     # diameter (mm) of each well on the plate
    depth=50)                       # depth (mm) of each well on the plate
[... 259 lines cut ...]
                    "diameter": 12,
                    "x": 18,
                    "depth": 75,
                    "total-liquid-volume": 0,
                    "y": 10,
                    "z": 0
                },
                "A6": {
                    "diameter": 12,
                    "x": 0,
                    "depth": 75,
                    "total-liquid-volume": 0,
                    "y": 50,
                    "z": 0
                },
                "B4": {
                    "diameter": 12,
                    "x": 6,
                    "depth": 75,
                    "total-liquid-volume": 0,
                    "y": 30,
                    "z": 0
                }
            }
        },`

lmtgalhardo, 2017-08-28:
For some reason, works normally in jupyter

astaff, 2017-08-29:
@lmtgalhardo can you confirm python versions in both environments?

lmtgalhardo, 2017-08-29:
running python Python 3.6.1 in jupyter, not sure how I would check what version of python the app uses but guessing 3 as well?

IanLondon, 2017-08-31:
confirmed in Python 3.5.1, opentrons version '2.4.2+195.g5b792ce'

But the JSON dict is unordered anyway, things like `my_container.wells()` are ordered correctly, it's just uglier to edit to JSON.

lmtgalhardo, 2017-08-31:
I had an issue with a customer where the wells were not functioning normally becasue of the json: if I said transfer to well 'A1', it would transfer to whichever well the unordered json had listed first. If I said transfer to well 'B1', it would transfer to whatever well the unordered json had as second, etc

lmtgalhardo, 2017-08-31:
Shown:
[image]
Above my output from jupyter, wells should be accessed in order A1 to D1, B1 to D1, C1 to D1

[image]

Shown above is a screenshot from a customer (testing here produced similar results) where the wells being accessed were the corresponding wells in that position of the unordered json

IanLondon, 2017-10-04:
The quick ugly fix for this to copy and paste this function into your protocol:

```python
def create_container_instance(name, grid, spacing, diameter, depth,
                              volume=0, slot=None):
    from opentrons import robot
    from opentrons.containers.placeable import Container, Well
    columns, rows = grid
    col_spacing, row_spacing = spacing
    custom_container = Container()
    properties = {
        'type': 'custom',
        'diameter': diameter,
        'height': depth,
        'total-liquid-volume': volume
    }

    for r in range(rows):
        for c in range(columns):
            well = Well(properties=properties)
            well_name = chr(c + ord('A')) + str(1 + r)
            coordinates = (c * col_spacing, r * row_spacing, 0)
            custom_container.add(well, well_name, coordinates)

    if slot is not None:
        robot.deck[slot].add(custom_container)
    return custom_container
```

After copying that function into your protocol, use it like this:
```python
plate1 = create_container_instance(
    '2x3_plate',        # name of you container
    grid=(2, 3),                   # specify amount of (columns, rows)
    spacing=(38.4, 38.4),          # distances (mm) between each (column, row)
    diameter=19,                   # diameter (mm) of each well on the plate
    depth=8,
    slot='A1'
    )                       # depth (mm) of each well on the plate

plate2 = create_container_instance(
    '2x3_plate',      # name of your container
    grid=(2, 3),                   # specify amount of (columns, rows)
    spacing=(38.4, 38.4),          # distances (mm) between each (column, row)
    diameter=19,                   # diameter (mm) of each well on the plate
    depth=8,                       # depth (mm) of each well on the plate
    slot='A2')                     # slot on robot to load container into
```

These container instances will function just like a the normal container instance returned from `containers.load`:

```python
p1000 = instruments.Pipette(
    axis='b',
    name='p1000',
    max_volume=1000,
    min_volume=100,
    channels=1,
    )

# plate1 and plate2 are just normal containers now
p1000.transfer(5, plate1.wells(0, to=10), plate2.wells(0, to=10))
```

Kalirren, 2017-12-04:
I just ran into this issue today.  What the robot though was Well 0 in a custom container was actually Well C2, well 1 was B5, etc. There really was no apparent pattern, other than that it was consistent across multiple instances of the container on the same deck.

I worked around it by getting the robot to map out which wells it was using. Then I created a dictionary to map the usual well numbers (0,1,2,3 ...) to the ones the robot understood.

BTW, this affects calibration, too.  If the robot thinks Well C2 is first, you have to calibrate the whole slot to the bottom of C2, not A1.