Opentrons/opentrons #319: container.rows length arg -> inconsistent result

IanLondon, 2017-08-30:
`length` argument gives inconsistent object types, which makes it harder to make length a parameter in a parametric protocol.

---

length >1 gives you a **WellSeries of WellSeries**

```python
> a = flat_plate.rows(0, length=2)
> a
>>> <WellSeries: <WellSeries: <Well A1><Well B1><Well C1><Well D1><Well E1><Well F1><Well G1><Well H1>><WellSeries: <Well A2><Well B2><Well C2><Well D2><Well E2><Well F2><Well G2><Well H2>>>
```

accessing / iterating -> gives you **WellSeries**

```python
> a[0]
>>> <WellSeries: <Well A1><Well B1><Well C1><Well D1><Well E1><Well F1><Well G1><Well H1>>
```

when length=1, you get a WellSeries, not a WellSeries of WellSeries. Accessing / iterating the result -> gives you Wells, which aren't iterable!!!

```python
> b = flat_plate.rows(0, length=1)
> b
>>> <WellSeries: <Well A1><Well B1><Well C1><Well D1><Well E1><Well F1><Well G1><Well H1>>

> b[0]
>>> <Deck><Slot C1><Container 96-PCR-flat><Well A1>
```

---

Slicing a list is kind of like setting the length. `[1,2,3][0:1]` gives you `[1]` not `1`.

So it would be more consistent and easier to use if length=1 gives you a WellSeries (or list) of a single WellSeries -- `[1]` instead of `1`.

IanLondon, 2017-10-17:
Similarly, `plate.wells([3])` returns a single Well, and `plate.wells([3, 4])` gives you a WellSeries of 2 wells.

As a result, you can get weird stuff like `[well.bottom() for well in plate.wells([list_of_wells)]`, which returns `[]` when list of wells has a length of 1 (because `[w for w in plate[0]]` returns `[]`.

```ipythonshell
In [4]: [well.bottom() for well in plate.wells([0])]
Out[4]: []

In [5]: [well.bottom() for well in plate.wells([0, 1])]
Out[5]: 
[(<Deck><Slot A1><Container 96-flat><Well A1>, (x=3.20, y=3.20, z=0.00)),
 (<Deck><Slot A1><Container 96-flat><Well B1>, (x=3.20, y=3.20, z=0.00))]
```

I'd expect `[well.bottom() for well in plate.wells([0])]` would give me `[(<Deck><Slot A1><Container 96-flat><Well A1>, (x=3.20, y=3.20, z=0.00))]`

IanLondon, 2017-10-17:
Also, I'd expect `plate.wells([])` would give me `[]` or maybe `None`, but instead it gives me a WellSeries of every well in the plate (so `plate.wells([])` currently is the same as `plate.wells()` )

This means if someone is writing a customizable protocol with a variable number of wells, they have to be very careful about whether a list of wells will contain 0, 1, or more items and use those conditions to call `plate.wells(list_of_wells)` or `plate.wells(0, length=num_of_wells)` differently for each case.

btmorr, 2019-07-29:
This has been resolved in APIv2, but won't fix in APIv1 as it would be a breaking change

closed 2019-07-29 by btmorr