Opentrons/opentrons #331: distribute() command in multi-channel pipette and WellSeries

astaff, 2017-09-06:
Protocol:

```
from opentrons import containers, instruments, robot

robot.reset()
plate = containers.load('384-plate', 'A1')

platerows = []
platerows.append(plate.rows(0).wells('A', length=8, step=4))
platerows.append(plate.rows(1).wells('A', length=8, step=4))

p50multi = instruments.Pipette(axis='a', channels=8)
matcell = platerows[0]

p50m.distribute(
    10, 
    platerows[0], 
    platerows,
    new_tip='once'
)

robot.commands()
```

Results in 20 aspirate/dispense pairs. Expected: 2 aspirate/dispense pairs.

Expected behavior can be achieved through:

```
p50multi.pick_up_tip()
for platerow in platerows:
    p50multi.aspirate(1.0, matcell)
    p50multi.dispense(1.0, platerow.bottom())
p50multi.drop_tip()
```

@andySigler

zubrabubra, 2017-11-15:
By default max_volume is set as min_volume + 1, thus in the case above we have max_volume set to 1, and min_volume value is 0. That leads the robot in situation to be able to transfer at max only 1ml of the liquid on each run.

```
Python 3.5.3
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from opentrons import containers, instruments, robot
>>> 
>>> robot.reset()
<opentrons.robot.robot.Robot object at 0x10e159f28>
>>> plate = containers.load('384-plate', 'A1')
>>> 
>>> platerows = []
>>> platerows.append(plate.rows(0).wells('A', length=8, step=4))
>>> platerows.append(plate.rows(1).wells('A', length=8, step=4))
>>> 
>>> p50multi = instruments.Pipette(axis='a', channels=8)
>>> matcell = platerows[0]
>>> 
>>> p50multi.distribute(
...     10.0, 
...     platerows[0], 
...     platerows,
...     new_tip='once'
... )
<opentrons.instruments.pipette.Pipette object at 0x10ef0fcc0>
>>> 
>>> robot.commands()
['Aspirating 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>',
'Dispensing 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Aspirating 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Dispensing 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Aspirating 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Dispensing 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Aspirating 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Dispensing 1.0 at ...
 'Dispensing 1.0 at <WellSeries: <Well A2><Well E2><Well I2><Well M2><Well A2><Well E2><Well I2><Well M2>>', 'Returning tip', 'Drop_tip ']
```

as you can see in example above we transfer bunch of times. But if we set amount of volume transferred to 1.0 (probably dot was missing):

```
Python 3.5.3
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from opentrons import containers, instruments, robot
>>> 
>>> robot.reset()
<opentrons.robot.robot.Robot object at 0x10b5c6f28>
>>> plate = containers.load('384-plate', 'A1')
>>> 
>>> platerows = []
>>> platerows.append(plate.rows(0).wells('A', length=8, step=4))
>>> platerows.append(plate.rows(1).wells('A', length=8, step=4))
>>> 
>>> p50multi = instruments.Pipette(axis='a', channels=8)
>>> matcell = platerows[0]
>>> 
>>> p50multi.distribute(
...     1.0, 
...     platerows[0], 
...     platerows,
...     new_tip='once'
... )
<opentrons.instruments.pipette.Pipette object at 0x10c386358>
>>> 
>>> robot.commands()
['Aspirating 1.0 at 
<WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>', 
'Dispensing 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>',
'Aspirating 1.0 at <WellSeries: <Well A1><Well E1><Well I1><Well M1><Well A1><Well E1><Well I1><Well M1>>',
'Dispensing 1.0 at <WellSeries: <Well A2><Well E2><Well I2><Well M2><Well A2><Well E2><Well I2><Well M2>>', 'Returning tip', 'Drop_tip ']
```

mcous, 2019-10-09:
Closing because this ticket is over 2 years old, is for PAPIv1, and I can't actually tell if there's a bug here

closed 2019-10-09 by mcous