Opentrons/opentrons #337: Check marks do not appear when saving calibration positions

jerrywan4, 2017-09-12:
Upon saving calibration data, check marks do not appear, making the app protocol unable to run the protocol. Occurs in version 2.5.2 and prior versions as well.

KarenCooke, 2017-09-19:
Recreated bug, attached are log files and code which does not save the last checkmark.

[otone-data-2017-09-19T09-31-13-04-00.zip](https://github.com/OpenTrons/opentrons/files/1314383/otone-data-2017-09-19T09-31-13-04-00.zip)

from opentrons import robot, containers, instruments

trough = containers.load('trough-12row', 'D2')
plate1 = containers.load('96-flat', 'B1')
plate2 = containers.load('96-flat', 'D1')
p200rack = containers.load('tiprack-200ul', 'B2')
trash = containers.load('point', 'B3', 'trash')

p200 = instruments.Pipette(
    name="p200",
    trash_container=trash,
    tip_racks=[p200rack],
    min_volume=20,
    max_volume=200,
    axis="a",
    channels=1
)

p200.distribute(55, plate1.wells('A1'), plate1.rows('2'))

andySigler, 2017-09-19:
The protocol you using in on axis `a` and is named `p200`. When I look that up in your calibrations file, I find the following plunger positions:

```json
"a:p200": {
            "positions": {
                "drop_tip": 28.0101,
                "blow_out": 27.0101,
                "bottom": 23.0101,
                "top": 8.0101
            },
```

Notice the `.0101` at the end of each position. The default plunger positions also have this (from `Pipette` line 119):
```python
# FIXME
default_positions = {
        'top': 0.0101,
        'bottom': 10.0101,
        'blow_out': 12.0101,
        'drop_tip': 14.0101
}
```

Then, inside `opentrons.utils.state.are_instrument_positions_calibrated()`, line 70 searches for that `0101` pattern to determine if a plunger is calibrated correctly.
```python
return not any([
    str(val).endswith('.0101') for val in positions.values()
])
```

There is a backstory to this and why it was added to the `2.5.x` release, essentially it is used to handle edge cases when calibrating pipette.

What seems to be happening here is that somehow these default values are (maybe) being inadvertently sent as coordinates to the smoothie board. Then if the user jogs the plunger with 1mm steps, then that `0101` decimal place will stay.

QUICK FIX:
Round all user-calibrated plunger positions to one decimal place. This will guarantee removal of those added decimal places, plus the user never even has access to the high of a resolution because the UI's jogging steps go down to `0.1mm` at the smallest.

here's a PR for that change: https://github.com/OpenTrons/opentrons/pull/342

astaff, 2017-12-20:
Closing. New calibration process in App 3.0 doesn't have this flow.

closed 2017-12-20 by astaff