Opentrons/opentrons #351: Make it easier to see tip tracking state

IanLondon, 2017-09-25:
Often when writing a protocol, I want to figure out how many tips I really need.

I might be missing an easy workaround for this, but since `Pipette.tip_rack_iter` is an `iter`, if I want to look how many tips are left, I can do `list(p200.tip_rack_iter)`. But by doing this, I've 'consumed' all my tips!

Is there an easy win for viewing tip tracking state without deeply changing how Pipette works?

https://github.com/OpenTrons/opentrons/blob/3e21da28a24cabe1f9dad4e80c781eb9ebfd4144/api/opentrons/instruments/pipette.py#L176

astaff, 2017-12-20:
Try:
```python

def get_tracking_state(pipette):
    from itertools import tee
    pipette.tip_rack_iter, tips = tee(pipette.tip_rack_iter)
    return len(list(tips))
```

if there is a use case for it in the API we can document it and make it a Pipette's property (cc @btmorr @LauraOpentrons )

IanLondon, 2017-12-20:
Cool! This might be a good item for an "advanced" section in the cookbook docs http://docs.opentrons.com/cookbook.html

sfoster1, 2019-12-13:
Maybe we'll add this to v2, but let's make a new ticket if so.

closed 2019-12-13 by sfoster1