Opentrons/opentrons #356: Feature request: prompt for user intervention

theosanderson, 2017-09-29:
It would be great to be able in the user interface to prompt for user intervention - e.g.

mcous, 2018-09-23:
As of `v3.4.0`, `robot.pause` now takes an optional `msg` parameter that will display in the run log of the app...

```py
robot.pause(msg='Pausing; please load tips')
```

...which will wait for the user to click "Resume". @theosanderson is the kind of functionality you were requesting?

(Although, while #2047 is still a thing, a pause message in the run log is of limited usefulness if it isn't correctly highlighted)

theosanderson, 2018-09-23:
Perfect - thank you.

b-cooper, 2024-03-18:
Just closing the loop on this now. Apologies for the delay in communication.

This was great feedback and we were able to incorporate it directly into the 6.3 release last summer.  Since then, `pause()` calls from the python api and any `waitForResume` protocol engine command will present a dialog box "front and center" along with the message that was passed as a parameter to the call. This experience is the same on OT-2 and Flex robots.  The latter will also provide a similar dialog on its embedded touchscreen display to catch the operator's attention.

[image]

closed 2024-03-18 by b-cooper