Opentrons/opentrons #360: (v3) RPC client doesn't support static class properties

mcous, 2017-10-02:
## overview

This issue is for:

- [X] Bug
- [X] App
- [X] v3.x.x
- [X] All OSes

## expected or desired behavior

An object with static class properties passed down the wire by the RPC server should be serialized by the client with that value as an instance property

## actual behavior

The RPC client assumes all properties of an instance's serialized `type` are methods of the instance, so it deserializes the static property as a method rather than a value.

@astaff checking whether a value of the `type` object is a primitive or object could be a path forward (in the type serialization, methods show up like `{method_name: {}}`, but then we hit the question of what to do with a static class property that is an object or array.

## steps to reproduce problem

This is currently observable in the session notifications, where the deserialized session remote has a method named `TOPIC` instead of a string property `session.TOPIC === 'session`

astaff, 2017-10-02:
`... should be serialized by the client with that value as an instance property` do you mean *deserialized* ?

Static class properties, known as attributes of class object in Python (https://docs.python.org/3.6/tutorial/classes.html#class-objects), are being passed as an attribute of a type, not instance and can be retrieved from server's control message upon initializing connection.

If we want to make it a property of an instance, we can explicitly set `self.topic = Session.TOPIC` or just `self.topic = 'session'`.

Also, i've added check for class object attributes being passed correctly in f7ae14f33f569f44fc0acf7ee74b52547b8e9a00

mcous, 2017-10-02:
> do you mean _deserialized_?

Yup, fixed.

> Static class properties, known as attributes of class object in Python (https://docs.python.org/3.6/tutorial/classes.html#class-objects), are being passed as an attribute of a type, not instance and can be retrieved from server's control message upon initializing connection.

Totally understood. The problem (and to be clear, it's a client bug, not a server bug) is that the client doesn't understand that not every name in the type object necessarily refers to a method name

> If we want to make it a property of an instance, we can explicitly set self.topic = Session.TOPIC or just self.topic = 'session'

No need for that sort of server change; I think the client should be fixed to handle this. I am curious, though, in reference to that client fix, if we'll expect more than primitive value to be present in a static class property

mcous, 2017-12-20:
I'm going to close this as a wontfix for now because I don't think it's particularly high-value. Will re-open if necessary

closed 2017-12-20 by mcous