Opentrons/opentrons #362: (v3) AttributeError in session.get_containers and get_instruments

mcous, 2017-10-02:
## overview

I'm unable to call `session.get_intruments()` and `session.get_containers()` via RPC, but I don't think it's RPC specific. It also may be related to the API setup problems I was having this morning (2017-10-02).

This issue is for:

- [X] Bug
- [X] API
- [X] v3.x.x
- [X] All OSes

## expected or desired behavior

Calling `session.get_containers()` and `session.get_instruments()` should return containers and instruments.

## actual behavior

Both methods error out for me:

- get_instruments: `AttributeError: 'Container' object has no attribute 'name'`
- get_containers: `AttributeError: 'Pipette' object has no attribute 'get_name'`

## steps to reproduce problem

In `app/ui/robot/api-client/client.js`, I added the following debugging lines to `handleApiSession`:

```js
  function handleApiSession (apiSession) {
    const {name, protocol_text, commands, command_log, state} = apiSession
    const protocolCommands = []
    const protocolCommandsById = {}

    apiSession.get_instruments()
      .then((instruments) => console.log('instruments', instruments))
      .catch((error) => console.error('error getting instruments', error))

    apiSession.get_containers()
      .then((instruments) => console.log('containers', instruments))
      .catch((error) => console.error('error getting containers', error))
    
    // ...
```

The errors will print in the log after uploading a protocol

astaff, 2017-10-02:
Fixed in 7e7b29820ebe932303fe46d568bdd9cee7ee7a6e please confirm

mcous, 2017-10-06:
Confirmed fixed

closed 2017-10-06 by mcous