Opentrons/opentrons #364: When I use `air_gap` arg, it does nothing, but gradually freezes my computer

IanLondon, 2017-10-03:
## overview

This issue is for:

- [x] Bug, feature request, or other (if other, please describe below)? **BUG**
- [x] API, App, both? **API**
- [x] Version? **2.5.0**
- [x] OS and OS version? **Linux Mint 18.1 Serena**

## expected or desired behavior

Don't crash when I use the `air_gap` arg :angry: :sob: 

## actual behavior

When I use `air_gap` in a transfer, and then run that .py with python on the command line, it gets stuck on that command. My computer slows down and eventually freezes.

## steps to reproduce problem

Run this in python:

```python
from opentrons import containers, instruments

p200rack = containers.load('tiprack-200ul', 'A1')
plate = containers.load('96-PCR-tall', 'C1')
trash = containers.load('trash-box', 'A2')

p200_multi = instruments.Pipette(
    axis="a",
    max_volume=200,
    min_volume=20,
    trash_container=trash,
    tip_racks=[p200rack],
    channels=8
)

print("looks good so far")
p200_multi.transfer(
    200,
    plate.rows(0),
    trash,
    air_gap=100  # if you comment this out, it runs fine. If left uncommented, it crashes
    )
print("this never prints")
```

That example is with an 8-channel -- I have the same problem with a single-channel, too.

zubrabubra, 2017-11-13:
@IanLondon i was not able to reproduce the problem. Do you think it's specific to Linux Mint 18.1 Serena?
Tried on mac os x tag 2.5.0 and branch v3a

astaff, 2017-11-13:
Unable to reproduce on OSX 10.12.6 and Ubuntu 16.06 with python 3.5.3 and 3.6.1. Tried against opentrons API PyPi `2.5.0`, `2.5.2` and current `v3a`

IanLondon, 2017-11-14:
@zubrabubra 
I think the problem was that my calibrations were corrupted (see #306 ), so it was silently using a pipette with a volume of 100. With a pipette of max vol 100 and `air_gap=100` you get the error you fixed in #421 !

btmorr, 2017-12-04:
@IanLondon can this issue be closed?

mcous, 2017-12-20:
Seems like we're unable to reproduce, so closing this issue

IanLondon, 2017-12-20:
I can still do

`p200.transfer(200, plate[0], plate[1], air_gap=1000)`

and get it to hang instead of throw an error.

It's not a linux-only, see https://github.com/Opentrons/opentrons/pull/421 -- infinite while loop

IanLondon, 2019-01-11:
Still open in legacy API, hopefully will be fixed in PAPI, but worth following up on

```python
from opentrons import labware, instruments

plate = labware.load('96-flat', 2)
p300rack = labware.load('opentrons-tiprack-300ul', 3)
p300 = instruments.P300_Single('left', tip_racks=[p300rack])

p300.transfer(200, plate[0], plate[1], air_gap=1000) # air gap way too high
```

this still does an infinite loop

sfoster1, 2019-12-11:
Happens on v2 also:

```
metadata = {'apiLevel': '2.0'}

def run(ctx):
    tiprack = ctx.load_labware("opentrons_96_tiprack_300ul", "1")
    right = ctx.load_instrument("p300_multi", "right", tip_racks=[tiprack])
    plate = ctx.load_labware('corning_96_wellplate_360ul_flat', '2')
    right.transfer(200, plate['A1'], plate['A2'], air_gap=1000)
```

PR #4588 (2019-12-11): fix(api): limit air_gap in transfer to <max volume
If you pass an air_gap larger than the pipette's working volume to any complex
command, in either v1 or v2, you get an infinite loop because it's subtracted
from the pipette's max volume to figure out how much can be aspirated in any
given transfer step.

Limit air_gap to between 0 and the pipette's expected working volume in transfer
and friends.

Closes #364
commit d7a3b700 (2019-12-11): fix(api): limit air_gap in transfer to <max volume (#4588)

If you pass an air_gap larger than the pipette's working volume to any complex
command, in either v1 or v2, you get an infinite loop because it's subtracted
from the pipette's max volume to figure out how much can be aspirated in any
given transfer step.

Limit air_gap to between 0 and the pipette's expected working volume in transfer
and friends.

Closes #364
closed 2019-12-11 by sfoster1