Opentrons/opentrons #365: (v3) Standardize internal datetime format

mcous, 2017-10-03:
## overview

We've got a few datetime formats floating around the API and app and I'd like us to pick one and stick with it (and apologize for my role in this mess).

This issue is for:

- [X] Bug, feature request, or other? **Other: chore-work**
- [X] API, App, both? **Both**
- [X] Version? **3.x.x**
- [X] OS and OS version? **N/A**

## expected or desired behavior

We should have a consistent internal datetime format. Given our codebase, I suspect the best option is probably milliseconds since epoch. @OpenTrons/py and @OpenTrons/js any thoughts?

## actual behavior

We use numbers and UTC strings in the API and app, respectively (I think).

## steps to reproduce problem

N/A

astaff, 2017-10-05:
Milliseconds since epoch sounds good to me

mcous, 2017-10-05:
Will make the switch on the app side during current `app-3-0-alpha` work on the app side. May mess up a few cosmetic things in the interim but that's not an issue

astaff, 2017-12-20:
@mcous can you specify where this is occurring and how things need to be changed to narrow down the scope of this issue?

mcous, 2017-12-20:
On the app side we store date as ISO strings in several places in the robot state, specifically with run time and session log handled stamps. I'm not sure how datetimes are stored in the API but added the API tag out of overabundance of caution + to suggest an audit

closed 2018-01-04 by btmorr