Opentrons/opentrons #370: Use Setuptools conventions in API instllation

astaff, 2017-10-05:
## overview

Let's use `pip install .` that uses `setup.py` to install the package and it's dependencies required to run the API and keep build/ci dependencies in `requrements.txt`. Refer to https://github.com/OpenTrons/opentrons/pull/347 for further discussion

- [X] Chore
- [X] API, Docker, Makefiles
- [X] Version 3.x

## steps to reproduce problem

This results in duplicating run dependencies in `setup.py` and `requirements.txt` to make non-compliant installation calls work.

mcous, 2017-12-20:
I would like to float [pipenv](https://docs.pipenv.org/) as a better dependency management solution. @btmorr has been playing around with it, and it looks like there's some Windows issues. It's use would probably be dependent on continued Windows build environment simplification

astaff, 2017-12-20:
Our python package dependencies seems to be in the right place. Run dependencies are in `setup.py`, development/ci deps are in `requirements.txt`, `compute`'s container uses `setup.py`. I haven't seen any windows issues with current setup. @btmorr let me know if there is anything we should improve.