Opentrons/opentrons #371: (v3) Memoize selectors with reselect

mcous, 2017-10-06:
## overview

During the alpha push, we've piled up a lot of selectors, especially related to tracking robot and calibration state. At the moment, these selectors are not memoized, which means we're doing a lot of duplicate calculations and generating a bunch of objects for the garbage collector. Memoizing the selectors with [reselect](https://github.com/reactjs/reselect) will reduce duplicate calculations and unnecessary intermediate objects

This issue is for:

- [X] Bug, feature request, or other? **Chore work**
- [X] API, App, both? **App**
- [X] Version? **v3.x.x**
- [X] OS and OS version? **All**

IanLondon, 2017-10-24:
:tada: woo `reselect` rocks, it also lets you compose selectors with less boilerplate :guitar:

IanLondon, 2017-12-22:
While we're at it, we may want to clean up robot selectors to make them single-labware-oriented and make "is robot moving now" and "is robot interacting with labware X now" selectors

PR #603 (2018-01-15): Memoize robot selectors and introduce flow to robot state
## overview

This PR was originally intended to address #592, but as I introduced reselect and flow to the robot selectors, I found out:

- The problem the ticket is trying to solve isn't really a problem
    - Or rather, I don't think we'll understand the problems we have until we start to implement #595
    - I'm going to hold off on #592 for now, because if I keep swinging blindly this PR is gonna get huge
- Flow is really helpful to have in the reducers and selectors, so I went a little hog-wild 

What I ended up with is a PR that:
 [... cut]
closed 2018-01-15 by mcous