Opentrons/opentrons #380: (v3) SessionManager.create emits a Session notification even if creation fails

mcous, 2017-10-25:
## overview

Currently, the API/app does not handle a rejected session creation due to the way the API sends out session notifications and the assumptions the app makes when receiving session notifications.

This issue is for:

- [X] Bug, feature request, or other? **Bug**
- [X] API, App, both? **API or App, depending on how we fix it**
- [X] Version? **v3**
- [X] OS and OS version? **N/A**

## expected or desired behavior

If I call `session_manager.create(...)` from the client and get an error, the app expects not to receive any subsequent session state change notifications, because as far as the app is concerned, no session exists.

We need to do one of two things:

* (API, my preference) Do not emit session notifications for sessions that we're not successfully created
* (App) Refactor app session handling to account for session notifications for sessions that failed during creation

## actual behavior

If `create` fails (for example, if the protocol passed to create had an error), the client receives the method call rejection (which is good) but then subsequently receives a session state change notification for the session that was not successfully created. The session notification it receives has errors in its `errors` array.

At the moment, the client is written in a way that assumes any session notifications are for active, real sessions, and that `errors` in those session notifications are for problems during a real protocol run. Because of this, receiving a notification for a "bad" session (as opposed to a "good" session that errors out during a run) confuses the client and throws the state out of whack.

## steps to reproduce problem

The best way to observe this problem is to watch the Redux log when you're doing this:

1. Connect the app to the API
2. Upload a protocol with syntax errors
3. In the log, observe:
    1. a `SESSION_RESPONSE` action comes in with `error` set to true (the create call rejection)
    2. Then, another `SESSION_RESPONSE` action come in with `error: false` (the session notification)

astaff, 2017-10-25:
Spoke to @mcous IRL:
1) API will reject method call and not send state change
2) API will also add protocol stack trace to error details

mcous, 2017-11-14:
This bug is still present if the protocol uploaded tries to use a non-existent container ID in container.load

PR #405 (2017-11-07): Don't emit Session notification even if creation fails
- fixes #380
- adds test simulating the behavior
- moves protocol load out of try-finally block
- tear down async test fixture gracefully
- fix existing test in light of new behavior
closed 2017-11-14 by astaff