Opentrons/opentrons #382: (v3, chore) Fix lint errors to get CI back to green

mcous, 2017-10-26:
`v3a` is currently [failing due to lint errors](https://travis-ci.org/OpenTrons/opentrons/builds/292767399). It would be lovely to fix those up and get our CI build back to green where it belongs

[image]

astaff, 2017-10-26:
api tests are broken as well with too much debug output to see the error

mcous, 2017-10-26:
> api tests are broken as well

Shoot, couldn't see that in Travis because the test command is:

```shell
python -m pylama opentrons tests && python -m pytest
```

Do we want to consider switching the test + lint task order in the API? On the app side we run test and then lint to make sure the more important thing (test failures) gets reported

closed 2017-11-01 by astaff

mcous, 2017-11-02:
Why was this issue closed? `v3a` CI is still red