Opentrons/opentrons #387: Unable to load new protocol after error

astaff, 2017-11-01:
# Expected

In the app, after loading a protocol containing an error, I expect to be able to fix the error and load the protocol again.

# Actual

1. Upload protocol with an error
2. Error screen is displayed
3. Load valid protocol
4. Screen is not changed and no message dispatched to the backend

Does it have anything to do with https://github.com/Opentrons/opentrons/issues/380 ?

mcous, 2017-11-02:
Yeah, this sounds very much related to #380, or at the very least difficult to diagnose until #380 is fixed. Are you going to have time to work on that or shall I take care of it?

astaff, 2017-11-02:
Looks like the user needs to change the screen after upload in order to upload again. Uploading, the switching to connect panel and then switching back seems to resolve the issue

mcous, 2017-11-02:
Able to confirm that we upload button only works once, outside of #380

mcous, 2017-12-11:
Fixed by #499

commit 16cf344f (2017-11-14): Fix #387 — don't emit state change if SessionManager.create errors out

- Removed adding error to log on _simulate, allowing
  exception to propagate instead

- Fixed test to fail if ANY state change is emitted

- Updated app to advertise local API instance as http
commit 9e5cc0f5 (2017-11-14): Merge pull request #428 from Opentrons/no-state-update-on-session-create-error

Fix #387 — don't emit state change if SessionManager.create errors out
closed 2017-12-11 by mcous