Opentrons/opentrons #403: arch not arching enough

cfblaeb, 2017-11-06:
## overview

This issue is for:

- [Bug] Bug, feature request, or other (if other, please describe below)?
- [API] API, App, both?
- [2.5.0] Version?
- [Ubuntu 17.10] OS and OS version?
 
## expected or desired behavior

Running the command:
p200single.distribute(5, my_chemical, my_container_thats_on_cold_deck.wells(0, length=8))
Should cause the p200single to pick up a pipette tip, pick up enough of my chemical and then move up the Z axis and then move via the XY to the colddeck and then down via Z. Aka Arch.

## actual behavior
It picks up a tip, it picks up my chemical, then it moves a little up the Z (enough that a tipless pipette doesnt hit the colddeck), then it moves in XY towards the coldplate where the tip collides with the deck. It then readjusts Z to the calibrated Z once it reaches its position.

Describe how the software actually behaves.

## steps to reproduce problem
1. Setup a p200 single with tips
2. Setup a chemical in some container.
3. Setup a 96 well plate on top of a colddeck.
4. Use the distribute command to move chemical into colddeck 96 well plate.
5. Notice collision.

IanLondon, 2017-11-10:
Thank you for raising this issue.

@cfblaeb When you have this problem, are you using jupyter notebook to run your script, or are you uploading a .py into the Opentrons App?

cfblaeb, 2017-11-10:
It happens both with jupyter and with app.
By reqeust from your support department I tried completely purging the installs (including settings in home folder) and reinstall, but it made no difference.
As a workaround I am now running robot.robot.arc_height = 100
It actually dont need to go up much higher than what the default arc function makes it go, but my plates are a bit high (maybe 1.5 times as high as "normal" 96 well) but well depth is "normal". My pipette tips are also "normal" height.
In general I guess pipette tip height could be estimated from calibrated pipette tip load position and then maybe the user should be able to make an optional calibration at top of container to ensure that pipette+tip arches over container top. Otherwise maybe just leave a bit more clearance room when moving over a container.
In fact collision with the magdeck when I have a slightly raised 96 well plate on it is common.

IanLondon, 2017-11-14:
@cfblaeb We're having trouble replicating the bug, could you please share a .py protocol that causes the collision?

cfblaeb, 2017-11-15:
I too appear unable to replicate anymore.
Between now and back when I could replicate it I have changed the deck layout and adjusted the pipette position (moved it further down create height differential with the multipipette).
I will close this issue and then re-open if I can once again reproduce it.

cfblaeb, 2017-11-15:
And now of course on second run, it collided with the cold deck =)
Absolutely nothing happened between the 2 runs.
Attached is the file.
`from opentrons import containers, instruments

tip200_rack = containers.load('tiprack-200ul', 'C3')
reagents = containers.load('tube-rack-2ml', 'B3')
coldplate = containers.load('96-PCR-flat', 'A1')
p200single = instruments.Pipette(
    axis='b',
    name='p200single',
    max_volume=200,
    min_volume=20,
    channels=1,
    tip_racks=[tip200_rack])
        
repairAtail_mm = reagents.wells('B2')

p200single.distribute(5, repairAtail_mm, coldplate.wells(0, length=8))`

mcous, 2019-10-09:
Don't know if this is still an issue. We should test the attached repro protocol and update this ticket accordingly

sfoster1, 2019-12-11:
We're currently confident in the height of arc moves in both API V1 and API V2. Please reopen this issue if the problem occurs again on 3.15 or later.

closed 2019-12-11 by sfoster1