Opentrons/opentrons #409: container.rows should be rows and container.cols should be columns

astaff, 2017-11-07:
# Current

```python
from opentrons import containers
plate = containers.load('96-flat', 'A1')
print(plate.rows(0))
```

outputs `<WellSeries: <Well A1><Well B1><Well C1><Well D1><Well E1><Well F1><Well G1><Well H1>>`

# Expected

`A1..A12`

LauraOpentrons, 2017-11-20:
If we invert them i,e plate.rows() is now A1-12 instead of A1-H1 that will create huge backwards compatibility issues for OT1 users and all past protocols written. Maybe we should have a meeting to discuss?

astaff, 2017-11-22:
Happy to discuss. We should keep in mind that for these kinds of changes we can automatically convert protocols. Also we should think about development experience for new protocols when writing a protocol we will be using rows instead of cols and vice versa.

astaff, 2017-12-20:
@IanLondon reassigning this to you since you are consolidating OT1->OT2 changes

PR #630 (2018-01-22): Issue 409 swap rows and columns

## overview

Fixes #409 , this PR swaps `Placeable.rows` and `Placeable.cols` to match the new container orientations.

Most notably, this changes which tips are picked up by a multi channel pipette, where before the multi would pick up `rows` of tips, now it picks up `columns` of tips.

## tested on machine 👍
closed 2018-01-22 by btmorr