Opentrons/opentrons #419: `distribute` with `touch_tip=True` always skips touch tip on last destination well

IanLondon, 2017-11-10:
## overview

This issue is for:

- [x] Bug, feature request, or other (if other, please describe below)?
- [ ] API, App, both?
- [ ] Version?
- [ ] OS and OS version?

## expected or desired behavior

When dispensing to A1, A2, A3, A4 with `touch_tip=True`, it should touch tip to all 4 destination wells before blowing out.

It probably shouldn't touch tip to the blowout container, but maybe in certain cases it should?

And, it should probably be optional to touch tip to the source container after aspirating?

Either way, it surely shouldn't skip tip touch the last destination well.

## actual behavior

It touches tip on all the wells but the last one. Blow out happens, then the touch tip follows the blow out:

```
'Picking up tip from <Deck><Slot A1><Container tiprack-200ul><Well C2>',
'Aspirating 170.0 at <Deck><Slot C2><Container trough-12row><Well A1>',
'Touching tip',
'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well A1>',
'Touching tip',
'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well B1>',
'Touching tip',
'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well C1>',
# NO TIP TOUCH ON C1!

'Blowing out at <Deck><Slot A2><Container trash-box><Well A1>',
'Touching tip',

'Aspirating 170.0 at <Deck><Slot C2><Container trough-12row><Well A1>',
'Touching tip',

'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well D1>',
'Touching tip',
'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well E1>',
'Touching tip',
'Dispensing 50.0 at <Deck><Slot C1><Container 96-PCR-flat><Well F1>',
# NO TIP TOUCH ON F1!

'Blowing out at <Deck><Slot A2><Container trash-box><Well A1>',
'Touching tip',
'Drop_tip at <Deck><Slot A2><Container trash-box><Well A1>'
```

## steps to reproduce problem

Use `touch_tip=True` arg for `distribute`:

`p200.distribute(50, trough[0], plate[:6], touch_tip=True)`

antonjs, 2018-01-16:
Just want to leave a +1 on this issue, and note that it's a significant problem for any protocol dealing with small volumes (like setting up a PCR). Without the touch_tip, the robot will reliably avoid dispensing things like primers into the final tube of a PCR strip :(.

mcous, 2019-10-09:
We have a repro protocol above. We should test it out and see if this is still an issue

IanLondon, 2019-10-10:
Same run log as Jan 11, tested with 3.13.1

khyox, 2019-10-15:
Reproduced with API 3.13.2:
```
( ... )
	.	[66] : Dispensing 25.0 uL into well G6 in "3" 
	.	[67] : Touching tip 
	.	[68] : Dispensing 25.0 uL into well G9 in "3" 
	.	[69] : Blowing out at well A1 in "12" 
	.	[70] : Touching tip 
	.	[71] : Aspirating 105.0 uL from well A1 in "11" at 1 speed
( ... )
```

It seems that the last touching and blowing steps (69 and 70, in this example) are wrongly swapped.
  
The consequence is an increase in the probability of dripping during the move from the last destination well to the blowout container, thus contaminating other wells. Unfortunately, we experienced such contamination.

sfoster1, 2019-10-16:
The fix for this issue will be released in 3.14.

PR #4231 (2019-10-16): fix(api): apiv1: touch tip before blowing out during transfers
A couple years ago, transfer, distribute, and consolidate were changed to always
blow out in the trash. Unfortunately, touch tip remained _after_ the blowout.
This meant that when a blowout was involved, the robot would dispense, then move
to the trash, then blow out, then touch tip (somewhere, somehow) in the trash.
This might lead to drips from the pipette as it moved between the destination
and trash contaminating wells in its path.

This commit fixes the issue by always blowing out, which involves moving to the
trash, after tou [... cut]
commit 294aa8fb (2019-10-16): fix(api): apiv1: touch tip before blowing out during transfers (#4231)

A couple years ago, transfer, distribute, and consolidate were changed to always
blow out in the trash. Unfortunately, touch tip remained _after_ the blowout.
This meant that when a blowout was involved, the robot would dispense, then move
to the trash, then blow out, then touch tip (somewhere, somehow) in the trash.
This might lead to drips from the pipette as it moved between the destination
and trash contaminating wells in its path.

This commit fixes the issue by always blowing out, which involves moving to the
trash,  [... cut]
closed 2019-10-16 by sfoster1