Opentrons/opentrons #461: App 3.0: Pipette with no numbers in name field

IanLondon, 2017-11-29:
## overview

This issue is for:

- [x] Bug, feature request, or other (if other, please describe below)? -- Bug
- [x] API, App, both? -- App
- [x] Version? -- v3a 3.0
- [x] OS and OS version? -- Linux Mint

## expected or desired behavior

I should be able to `name` my pipette any random thing, and its max volume should come from `max_volume` field.

## actual behavior

The app uses the pipette's `name` attribute to determine the pipette's volume or model. This happend in fn `apiInstrumentToInstrument`.

It does regex on something like `'p200'` to get `200`: `const volume = Number(name.match(RE_VOLUME)[1]);`

If I don't name it with the volume in the name (including leaving the default name, `'Pipette'`), then the regex returns nothing, so trying to do `[1]` on the result throws the error: `Uncaught (in promise) TypeError: Cannot read property '1' of null`

## steps to reproduce problem

Define pipette without name matching the `/.*?(\d+).*?/` pattern

```python
pipette = instruments.Pipette(
  max_volume=200,
  trash_container=trash,
  mount='right',
  # name='p200',  # <-- uncommenting this restores function in app
  tip_racks=[tip_rack]
  )
```

then upload that protocol to the app

astaff, 2017-11-29:
If the volume can't be inferred from the name or `max_volume` I would suggest leaving it blank in the UI instead of not loading the protocol at all

mcous, 2017-12-20:
Desired behavior is for the app to pull the tip size from the tiprack that it interacts with. At the moment, we know in the app which tiprack the pipette touches. However, the only way we can know what size tip a given tiprack has is by inferring it from the type (e.g. `tiprack = containers.load('tiprack-200ul')`), which kinda leads to the same issue

IanLondon, 2017-12-20:
It's not a very common use case, but it is possible users will use multiple tip sizes with the same pipette. Eg a p300 can use 100uL, 200uL, and 300uL tips. However, I cannot `containers.load('tiprack-100ul', slotName`. 

Something to think about is that tips can be pretty much any uL size -- eg there are 80uL tips and 50uL tips, though they aren't very common.

The problem I think we're trying to solve is that tips are different lengths, and that affects calibration. So we need a clear way for a user to associate tip model with protocol.

When I'm writing a protocol, I want to specify 100uL tips, because that will give me different accuracy than 300uL tips and change how the liquids move during the protocol.

Also, I might use Brand A 200uL tips, and next year my lab switches to Brand B 200uL tips, which are a different length -- I need to recalibrate.

Lastly, I might switch between 200uL filter tips of X mm and 200uL non-filter tips of Y mm within the same protocol.

mcous, 2018-04-26:
I'm fine calling this issue closed and superseded by the addition pipette config to `labware-definitions`, the default pipette constructors with fixed names, and #1260

closed 2018-04-26 by mcous