Opentrons/opentrons #526: Robot wifi credentials cannot be set from the app

btmorr, 2017-12-20:
## overview

Users should be able to configure their WiFi credentials from the app

## behavior

**Current behavior**: In order to set WiFi credentials, you have to ssh into the robot and modify a file

**Desired behavior**: The user should be able to set up WiFi credentials from the app, after connecting to the robot with the USB cable.

Kadee80, 2017-12-20:
I have (a wildly deprecated) starter branch for this in the UI.  Happy to resurrect this work to get something testable for connect.

mcous, 2017-12-21:
Similar comment to the one I left in #528: I'd collected the following acceptance criteria for this story:

* API should provide an endpoint to set WiFi credentials
* App should provide interface for UI to send WiFi credentials to that endpoint
* API should provide interface type for a given connection
    * e.g. in `/health`, the API could add interface: 'wired' or 'wireless' or similar depending on which interface the request came in on
    * This is a nice to have; app can probably infer this based on current connectivity strategy
* App should differentiate between wireless and wired robots
    * Related: app should order robots based on priority of: currently connected, wired, wireless

btmorr, 2018-01-04:
In terms of implementation, these settings are stored in a file in Resin host OS that is not accessible to the container. 

`nmcli` can be used to modify the settings, so it should be straight-forward to make an endpoint that takes { "ssid": "<network_name>", "psk": "<secret>" } and sets using something like `nmcli connection modify resin-wifi-01 802-11-wireless.ssid <network_name> 802-11-wireless-security.psk <psk>`. I haven't figured out the exact syntax of the command, but it will look something like that (don't know if you can set more than one property at once, for instance, or if it has to be two separate commands). See https://developer.gnome.org/NetworkManager/stable/nmcli.html

umbhau, 2018-01-30:
Any objections to moving the FE requirements in #697 and #698

mcous, 2018-02-06:
Closing in favor of tickets above

closed 2018-02-06 by mcous