Opentrons/opentrons #528: Check for app updates and notify user if update is available

btmorr, 2017-12-20:
## overview

The app should check for updates and notify the user if an update is available

## behavior

**Current behavior**: 

- The user must check if a new version of the app is available, and download manually. 
- They must also manually push server code to the robot.

**Desired behavior**: 

- The app should check if a new version of the app is available, and prompt the user to download and install it. This should install the new version of the app. 
- When the app connects to a robot, it should check the version of the server running on the robot. If that version is lower than the version of the app, the app should prompt the user whether or not to update the software on the robot. If they say "Yes", the app should download the server code and push it onto the robot.

btmorr, 2017-12-20:
Waiting on #527

mcous, 2017-12-21:
Do we want to consider splitting this into multiple tickets (or generating a checklist)? I think front-end vs back-end update would be reasonable. Based on when I was scoping this out, I saw the following acceptance criteria for "updates":

**Front-end**

* App should be able to determine if itself is out of date
* App should prompt the user to update itself if out of date
    * Prompt should include release metadata (e.g. changelog)

**Back-end**

* API should make versions available to the app
    * API version
    * Compute version
    * Smoothie version
* App should prompt user to update if any backend versions are out of date
    * Prompt should include update metadata
    * Could be a ticket per backend component
* API should accept a new version of the API from the app to install and run
* API should accept a new version of smoothie to install and run
* API should allow the user to opt-in or out of a compute update
    * Mechanism TBD and likely needs its own ticket
*

pantslakz, 2018-01-19:
Do we want to allow users the option to revert to a prior version?

umbhau, 2018-02-12:
@mcous I like your suggestion to split this into FE and BE stories. 

I'll split out the FE work into separate ticket(s). @btmorr, as far as BE work, #641 exposes API and FW versions, but I can't find any previous updates adding a feature for the API to accept new versions of the API and/or Smoothie to install and run. Could you confirm we still need this (and I'll create a ticket)?