Opentrons/opentrons #530: Database migration is slow, and can get out of sync with containers.json

btmorr, 2017-12-20:
## overview

Database migration when container db does not exist takes a long time. Also, the db can get out of sync with the json file, which leads to confusing bugs.

## behavior

**Current behavior**: When the server starts, if the containers db does not exist, the data from containers.json is translated into the db. Once that db exists, if a container definition is changed, the change is only in the db, and modifications to containers.json do not cause an update of the db.

**Desired behavior**: There should be only one place where container definitions are stored, and that source should be more efficient in terms of read/write speed and memory use

btmorr, 2017-12-20:
Suggested solution: remove the db altogether, and break the containers.json file into an individual json file for each container. These individual files could be read/updated directly, and only the ones in use would be read, which would be faster and have a smaller memory imprint.

IanLondon, 2017-12-21:
We also talked about:
* removing the x and y offsets, since they're just an extra math step when loading and don't mean anything
* moving containers to its own project in the monorepo

And we should probably flip x & y, to match our orientation

btmorr, 2019-07-16:
Will be closed by #2187

sfoster1, 2019-07-19:
Closed by #3713

closed 2019-07-19 by sfoster1