Opentrons/opentrons #531: Button on robot frame doesn't do anything

btmorr, 2017-12-20:
## overview

The button on the frame is available, and should ostensibly have some function.

## behavior

**Current behavior**: nothing

**Desired behavior**: something

btmorr, 2017-12-20:
I haven't heard complete consensus as to what this button should do. The most common behavior I've heard mentioned is to halt the robot, but I haven't seen anything formally specifying this.

IanLondon, 2017-12-20:
+1 for emergency halt during run

There was talk of a "replay" behavior when the robot is idle.

If the robot is paused during a protocol run (as in "pause until user hits resume"), maybe it can resume there, mimicking the behavior of "Resume" in the app?

@andySigler  ???

andySigler, 2018-01-17:
Here's a minimal flow Jackson and I made way back last spring

[image]

pantslakz, 2019-01-31:
@mcous @btmorr would be lovely to get the above on the robot sometime soon. Morgan and I had the chance to work through this with Andy and Jackson - looks solid!

howisthisnamenottakenyet, 2019-05-07:
I edited the description to add some design considerations

Matt-Zwimpfer, 2021-06-03:
MHH - let's make a decision; @readywater has been considering this

readywater, 2021-06-28:
[image]

Basic pause/play and cancel interactions from the button. Would also suggest adding a log object with timestamp and interaction point (app/robot/api) the pause/play/cancel steps are triggered.

readywater, 2021-07-01:
Adding a quick revision to this:
**Pause** functions should aim to be non-destructive, since pausing is often used to correct a human or minor robot error. Pausing should retract Z axis to allow for manipulation of labware on the deck. 
If moving, along X or Y, movement should be completed and Z axis should remain retracted (eg. move to dispense location, but don't dispense)

**Cancel** should be immediate, with the idea that canceling an action is to prevent some kind of physically destructive. After canceling, the robot should be homed from a software client.

emilywools, 2021-07-08:
hey @readywater question for you:

"if the protocol is canceled from the robot, the post-confirm cancelation experience should start in app"

Does that mean the run is not actually cancelled if the user presses + holds the physical button until the user confirms in-app?

readywater, 2021-07-12:
@emilywools No, the run is canceled by press and hold without in-app confirmation. The app should still have a confirmation dialog though.

mattwelch, 2021-07-13:
We'll write a new implementation ticket to execute on the outcomes noted on this ticket.

closed 2021-07-13 by mattwelch