Opentrons/opentrons #532: Drop tip button in tip box calibration crashes robot constantly

rvinzent, 2017-12-20:
## overview

When calibrating tip boxes, the drop tip option on boxes after the first one will return the tip to the first box and crash the robot.

## behavior

Expected: 
-robot should return the tip to the same box it was pulled from. 

Actual:
-robot returns tip to the first box regardless of the box it came from.

## steps to reproduce

simply calibrate more than one tip box and press the drop tip button when calibrating any box that isn't the first. **This crashes the robot every single time the button is pressed and desperately needs a fix ASAP**

btmorr, 2017-12-20:
Hi! Thanks for reporting this. Which version of the app are you using, and what is your operating system?

rvinzent, 2017-12-21:
I have app version 2.5.2 and OS is Windows 7 Enterprise

pgladkevich, 2018-02-05:
To update this issue as I work in the same lab as Ryan: The robot returns the tip to the tip-box in lowest position alphabetically first, and highest position numerically second. So if there are two tip boxes, one in the A2 slot and one in the B3 slot the robot will attempt to drop or pickup tip from only the A2 slot. Similarly if there are tip-boxes in the A3 and A2 slots only the A3 slot is accessible unless you use MOVE TO.

mcous, 2018-05-17:
Closing this ticket because incorrect behavior during tiprack calibration was fixed for the [OT-2](http://opentrons.com/ot-2) and `v3.0.0` of the [app](http://opentrons.com/ot-app). Unfortunately, given how different the OT-2 is to the OT-1, we don't have the engineering resources available to backport fixes to the legacy app.

That being said, if anyone from our community would like to make updates to the legacy app, we're happy to re-open tickets and help out. The legacy version of the app lives in the `ot1-stable` branch of this repository.

closed 2018-05-17 by mcous