Opentrons/opentrons #533: Sphinx-generated API documentation should reflect the user-facing API, not the library internals

IanLondon, 2017-12-20:
Documentation request:

In our API's base `__init__.py`, instruments and containers get wrapped to be called with the robot singleton that is also created in that file, so the `robot` arg to our constructor functions is masked to the API user.

Our Sphinx-generated documentation doesn't account for this wrapping, so the user-facing API (what you use when you do the canonical `from opentrons import instruments`) differs from the API documentation -- there's no `robot` arg in `instruments.Pipette` when you do the canonical import.

At http://docs.opentrons.com/api.html#module-opentrons.instruments we have:

`class opentrons.instruments.Pipette(`**robot**`, axis, name=None, channels=1, min_volume=0, max_volume=None, trash_container=None, tip_racks=[], aspirate_speed=300, dispense_speed=500)`

NOTE: This example is from v2.5.2 but I think that the wrappers that hide `robot` will cause the same problem for our Sphinx API docs.

IanLondon, 2018-07-12:
still true

sfoster1, 2019-12-13:
Not anymore really. Or at least the extent to which it does is limited to v1.

closed 2019-12-13 by sfoster1