Opentrons/opentrons #541: Required to import unused modules

rvinzent, 2017-12-21:
## overview

Scripts require you to import some unused modules to function. I tried to run a simple MagBead text with the Mag module by importing only the bare minimum. The script uploads without error but `Run` is unclickable and no exceptions or warnings are raised. Issue is fixed when I import `containers` , `robot', and `instruments` and register a pipette (never used). I suspect the `robot` module must be imported to function, but I don't have time to isolate this issue. This is undocumented and quite confusing to understand why the app won't run a 3 line protocol.

## behavior

## steps to reproduce

mcous, 2017-12-21:
Hi, thanks for your issue report! Would you mind pasting the contents of your protocol?

rvinzent, 2017-12-26:
Code:

    from opentrons import instruments

    mag = instruments.Magbead(name='mag')
    mag.engage()

Adding pauses or disengage does nothing for me. `Run` is unclickable although the protocol was uploaded successfully and there are no calibrations to perform. My initial thought was incorrect as importing the `robot` module doesn't seem to make a difference.

mcous, 2018-05-17:
Hi! This seems like an issue with the OT-1 / the previous version of the app and API. Given that we're unable to devote resources to our legacy software, I'm going to close this ticket for now.

We're aware of the difficulty of working with many aspects of our Python API, these sorts of issues with `import`s included. We're actively working to pay down this technical debt in our API to make things behave in a more canonical and expected manner from a software context. If this specific issue still turns out to be a problem with the OT-2, we'll definitely re-open this ticket or point you to the ticket(s) that will address it.

closed 2018-05-17 by mcous