Opentrons/opentrons #543: Address connectivity inconsitencies, especially cross-platform

mcous, 2017-12-21:
## overview

We're experiencing a number of connectivity issues, some of which could be easily mitigated with a few changes in the app and API.

## behavior

* User is unable to establish a connection, especially on Windows
    1. User clicks the connect button
    2. Behind the scenes, the connection fails
    3. The app doesn't indicate that the connection failed
    4. User is confused and unhappy
* Connection drops unexpectedly (read: non-gracefully) and the user is not notified
    1. API errors out (or similar) and socket it not properly closed
    2. App doesn't react because socket never emits a `close` event
    3. User is confused and unhappy
* User unplugs robot and app does not respond (especially on macOS)
    * Similar experience to above, especially on macOS, because socket doesn't always emit `close` on cable yank

## reasons and proposals

For (1), we can do two things:

- [ ] Connect via IP address rather than MDNS hostname
    - Windows connections can fail because `whatever.local` will not resolve unless bonjour is installed
- [ ] Display an error message if a connect request fails
    - App already has all the information it needs for this, it just needs to act on it

For (2) and (3):

- [ ] App and API should implement a manual ping <> pong scheme to detect frozen sockets quickly
    - App already has a simple alert for unexpected socket `close`s
    - A ping-pong could get us to that useful alert sooner

mcous, 2018-01-03:
> Display an error message if a connect request fails
>     * App already has all the information it needs for this, it just needs to act on it

@pantslakz do we have any assets for displaying this sort of error?

1. User clicks button
2. Button text changes to "spinner" of some sort?
3. Error occurs
4. Message displays somewhere?

umbhau, 2018-01-30:
@mcous any objection to pulling the FE requirements out of this ticket and into #694 and #695?
" Display an error message if a connect request fails"

mcous, 2018-01-30:
Closing in favor of the tickets above!

closed 2018-01-30 by mcous