Opentrons/opentrons #553: App should perform deckware calibration using "correct" pipette

mcous, 2017-12-22:
## overview

**Bugfix + Feature request**

Currently, app v3 does all deckware calibration with its attached single-channel pipette. If no single-channel is available in the protocol, the app crashes hard (**bug**). This blocks usage and testing of multi-channel-only protocols.

Really, the "correct" pipette to use for deckware calibration is a pipette that actually interacts with the deckware during the protocol (**feature**). This feature is slightly blocked by some necessary cleanup work in the app's calibration state, and is also blocked by UX.

## behavior

### current

If you load a protocol without a multi-channel, the app errors out to a blank screen when you try to go to the calibration screen.

If a single channel is available, the app always requests the API to use the single-channel pipette to calibrate deckware.

### desired

#### stage 1

- [ ] **Stage 1**
    - If you load a protocol without a single channel, the app should just fallback to the multi-channel
    - Certain assets won't be correct, but this is _way_ better than current behavior
    - Not blocked by UX
    - Not blocked by API
- [ ] **Stage 2**
    - Deckware calibration should be performed with a pipette that interacts with that deckware
    - Prefer single channel, support multi channel
    - Blocked by UX (how should tips and tipracks be handled?)
    - Blocked by API (probably need tip tracking for this)

jerrywan4, 2017-12-27:
> current
If you load a protocol without a multi-channel, the app errors out to a blank screen when you try to go to the calibration screen.

I believe this should say 

> current
If you load a protocol without a **single channel**, the app errors out to a blank screen when you try to go to the calibration screen.

pantslakz, 2017-12-28:
First round of UI designs for **Stage 2** will be shared end of week next week (Jan 4th or 5th), with aim of getting feedback - and delivering refined designs by Wednesday January 9th.

LauraOpentrons, 2018-01-02:
To elaborate more on point two for labware calibration:

Tip probe works for multi-channel, however, it will not use the mutlichannel to calibrate any other labware.

mcous, 2018-01-04:
Stage 1 landed in #572

pantslakz, 2018-01-10:
@mcous @andySigler if a protocol has both a single and a multi channel pipette - will it default to calibrating with single? Or will it calibrate using the pipette that's interacting with the labware user is calibrating? 

> Really, the "correct" pipette to use for deckware calibration is a pipette that actually interacts with the deckware during the protocol (feature).

Trying to understand if this mean whether calibration could happen with both Pipettes attached (ie. one single and one multi), or if it will always default to calibrating all containers using just one pipette (specifically single pipette, if available)

mcous, 2018-01-10:
As we discussed over the phone, from a technical standpoint we don't care which pipette is used to confirm the position of given labware. We have all the offsets thanks to deck calibration + tip probe, so this can be purely a "what's least confusing for the user"

pantslakz, 2018-01-12:
Following up to comment above - realized yesterday that we *do* care which pipette is used to confirm positions of tipracks, but we *don't* care which pipette is used to confirm other labware.

Some tips will not fit on certain pipettes, and for this reason calibrating a tiprack must be done with the pipette associated with it. Additionally this means that at this time, each tiprack is associated with only one pipette. @mcous does this possibly mean we can remove the API block mentioned above?

mcous, 2018-01-23:
Closing in favor of #621

closed 2018-01-23 by mcous