Opentrons/opentrons #554: Need defined names for loading in containers for a particular deck slot

LauraOpentrons, 2017-12-22:
## overview
The current deck slot naming convention is confusing and does not correspond to deck numbers.

## behavior

Currently in v3a, you can only access the slots on the robot as 'A1..A4', 'B1..B4', 'C1..C4'.

Would like to be able to access the slots either using strings with the values '1'...'12' which correspond to the engravings on the robot or with integers also going from 1...12.

## steps to reproduce

N/A

IanLondon, 2018-01-02:
Cool. Let's be sure to update the docs with the PR that fixes this too!

Though we will support both strings and numbers (`'1'` and `1`) in the API, I suggest we should just choose one to be "canonical" for the docs - so all the examples that use `containers.load` to get a container and do something with it, but aren't examples of the `containers.load` command, will be uniform.

(This would be like how myPlate.wells(...) is the "canonical" way to get wells in the docs, though you can also do myPlate[:3] etc)

IanLondon, 2018-01-19:
#631 fixes this, but does not update the docs

closed 2018-01-22 by btmorr