Opentrons/opentrons #559: Calibrate trough:A1 center with single channel, and multi-channel crashes

andySigler, 2017-12-28:
## overview

When calibrating with `v3a`, the user is guided to locate a center (either top or bottom) of a trough's first well using an attached single-channel pipette. That same "center" position is then used to align with the multi-channel's rear-most tip.

## steps to reproduce

Load protocol that uses a `trough-12row`, and both a single-channel and multi-channel pipette. Calibrate the trough's position with the single-channel, then run the protocol. When the multi-channel attempts to go into any of the trough's wells, it's front-most tips will collide with the container.

## current work-around

When calibrating a trough with single-channel, align the tip with the spot on the trough where a multi-channel's rear-most tip would be. This allows the multi-channel to safely enter the trough, but creates an inaccurate picture of the deck's layout.

mcous, 2017-12-29:
> When calibrating with `v3a`, the user is guided to locate a center (either top or bottom) of a trough's first well using an attached single-channel pipette

While the text says "accurately centered" (which is definitely misleading), the hover diagram shows the correct location:

[image]

Is this ticket for changing the text, or is it for updating the API to allow the user to calibrate to the center of the tough with the single channel? Also, addressing #553 could help in the case where the trough is used only by the multi-channel

andySigler, 2018-01-02:
@mcous ok thanks I'll remove this from the milestone, but I'm leaving the issue open until we put more thought into it. This behavior is a side affect of hour the API currently works and is confusing, let's scrutinize this a bit @pantslakz

mcous, 2018-02-08:
Closing this ticket in favor of #783, which adds clearly visibly diagrams with correct trough alignment depending on pipette type

closed 2018-02-08 by mcous