Opentrons/opentrons #568: Switch to `yarn` for JS projects

mcous, 2018-01-03:
> This is a Request for Comments ticket (RFC, see pretty new label). I think this could be a pretty good way of collecting thoughts for proposals and whatnot so we can keep everything close to our ticketing (and also out in the open, so potential contributors can see how things are decided). Feel free to chime in with RFC process stuff in this ticket as well!

## overview

After playing around with monorepo organization stuff over the break, I think I'm finally ready to pull the trigger on switching from `npm` to `yarn` for package management for our JavaScript dependencies.

Pretty sure everyone already knows, but just in case: `yarn` is an alternative package manager to `npm` it uses the same package registry (so everything on npmjs.org is installable with `yarn`) and the same `node_modules` installation directory (so node and webpack continue to function normally).

### problems we've been having

In no particular order:

* Windows dependency installation bugs
* Windows package linking bugs (because symlinks aren't the same as junctions)
* Weird lockfile churn, especially with cross-platform optional dependencies
* A bunch of duplicated installs of packages that are used in all our JS projects, like React

## benefits

* [Workspaces](https://yarnpkg.com/lang/en/docs/workspaces/)
    * Yarn has built-in support for muti-project repos
    * This means only one install of shared dependencies and only one lockfile
    * **Most usefully** it will automatically link interdependent projects (great for components library)
    * Will require some work to ensure no dependency version conflicts
* Faster installs across the board
    * Even if `yarn` didn't gracefully handle shared dependencies, installs are still faster with `yarn` than `npm`
    * Offline installs from your own package cache are well supported, if that's a thing you care about
* `npm` v5 has been all sorts of buggy for us (especially with Windows)
    * Switched to v5 for lockfile support, which is built into yarn
    * Every single bug we've encountered had an issue thread on `npm` with a bunch of comments reading "I switched to yarn and this problem went away"
* Well supported (Facebook project)

## downsides

* Switching costs
    * Will need to redo all our lockfiles, Makefiles, etc.
* Younger tool, which can mean exciting new bugs that we're not expecting
    * `npm` being older hasn't seemed to prevent a bunch of bugs
    * Hopefully, Facebook's support means bugs get squashed quickly
* Extra tool users will need to have installed

IanLondon, 2018-01-03:
Yarn recommends you don't use npm to install yarn https://yarnpkg.com/en/docs/install#alternatives-tab so we gotta figure out how we want to install yarn on mac and windows builds... does the mac build environment have homebrew already?

mcous, 2018-01-03:
According to the [CI page in the Yarn docs](https://yarnpkg.com/en/docs/install-ci)

* `yarn` is pre-installed in AppVeyor
* Travis will pick up the presence of a `yarn.lock` file in repository root and run `yarn install` automatically
    * If this doesn't work, there's a shell script installer for Unix

We're not running macOS CI at the moment, but when we were, we used `brew`, so that would also be a path if needed

mcous, 2018-01-04:
Tracking implementation in #576

closed 2018-01-04 by mcous