Opentrons/opentrons #570: Basic Protocol Designer Action Functionality

IanLondon, 2018-01-03:
# WIP/draft! saving now so I don't lose this

## overview

This is an outline of how Action creation and editing will work in Protocol Designer, to inform how the engineering team should design Redux actions for PD. (I'll capitalize PD "Actions" to distinguish then from "Redux actions")

## behavior

Initially, when you get the the "main PD editor page" (Gear icon in Nav), PD starts out with an empty deck and has one Action, "Labware & Ingredient Setup", in the Actions List.

You can't delete the initial "Labware & Ingredient Setup" Action. (right?)

Zero or 1 Action is selected at any given time (there's no current design for what happens when zero actions are selected, but Morgan would like that state available).

When the "Labware & Ingredient Setup" Action is selected, the Deck acts as it did in the initial labware & ingredient setup prototype we built before. Labware slots allow you to add labware which then get a name, and occupied slots have hover states that let you Edit Ingredients and Remove Labware. There's also a way to Copy labware but the details are TBD.

By clicking "+ Add Action", additional Actions can be added to the Action List. When a new Action is created, it automatically becomes the selected action.

When any Action besides "Labware & Ingredient Setup" is selected, the Action Editor section appears showing settings for the selected Action.

Most Actions have an "Edit Selection" button that opens a Well Selection Modal when clicked. This is paired to a dropdown menu to select the labware instance. When no corresponding labware is selected, the "Edit Selection" button doesn't do anything.

In the Well Selection Modal, the user can select wells, then click the "Save Selection" in the Well Selection Modal. This assigns a list of wells `['A1', 'A2', 'B1', 'B2']` to that part of that action (eg, to `sourceWells` / `destWells`).

# Considerations for later

Deleting actions can have cascading effects on downstream actions. For example, if you delete an action that fills an empty well with Ingredient X, then a later action aspirates from that well, now you're aspirating nothing. We will need a way to notify the user that deleting an action effects some downstream actions. There

Kadee80, 2018-01-04:
RE: `Current "page": the pages that exist are still TBD, but there's an idea that the Nav on the leftmost side of the screen moves you across different pages.`
Not terribly familiar with the PD flow, but the way it works in the app is leftmost `NavBar` changes `SideBar` content. `Sidebar`  actions/routes change larger `task` or `page` panel on the right.

IanLondon, 2018-01-05:
https://github.com/Opentrons/opentrons/milestone/2 addresses prework, NavBar, SideBar, and Step creation/deletion. It does not address anything happening on the right side of the screen, that stuff will be left to later milestones.

closed 2018-04-26 by IanLondon