Opentrons/opentrons #591: Enable style exports from Component Library to other projects

Kadee80, 2018-01-09:
# Overview

Currently the app, protocol designer, and component library are utilizing CSS modules for styling. The component library currently allows for component sharing  via `@opentrons/components` but does not yet allow for CSS styles and values/vars to be shared across projects. This results in  multiple overlapping style sheets/variable definitions within each project. At the very least, all color and typography variables and atomic css styles should defined within the Components Library and exportable to App PD and any other outside projects where OT Branding and Components are utilized.

## Currently with CSS Modules:
Not able to use css `@import`. This matters because `colors.css` and `typography.css` can not be aggregated to index.css for exports from `@opentrons/components`
`@values` require individual imports within each scoped css file
loss of global css definitions
`composes` will not work with compound selectors.

## CSSNext Implementation as possible solution:

1- Utilize `postcss-import` module. (Note: In past tests, this module did not work with CSS modules and @values).

This would enable:
```css
@import 'styles/colors.css';
@import 'styles/typography.css';
@import ...
```
from index.css, allowing for a single file export for all sharable styles, vars etc.

2- Implement [postcss-css-next](http://cssnext.io/postcss/) in component library. Replace @value imports with root variables.

#### Currently with CSS Modules:
```css
@value color_light_grey, color_med_grey, color_dark_grey from '../styles/colors.css'
```
#### With CSS Next:

colors.css
```css
  root: {
    --light-grey: #CCC;
    --med-grey: #999;
    --dark-grey: #333;
  }
```

index.css
```css
@import './styles/colors.css';
```

some-component.css
```css
  .btn {
    background-color: var(--med-grey);
  }
```

3- Use CSS Next variables for atomic styles rather than CSS Modules' `composes`

#### Currently with CSS Modules:

typography.css
```css
  @value color_font_dark: #4a4a4a;
  @value font_size_huge: 3rem;
  @value font_weight_bold: 800;

  .font_huge_dark {
    font-size: font_size_huge;
    font-weight: font_weight_bold;
    color: color_font_dark;
  }
```

some-component.css
```css
  .title_bar {
    composes: font-huge-dark from '../styles/typography.css'
  }
```

NOTE: Below compound selector will not work v
```css
    .title_bar .rounded_corners{
    composes: font-huge-dark from '../styles/typography.css'
  }
```

#### With CSS Next:

typography.css
```css
  root: {
    --font-dark: #4a4a4a;
    --font-huge: 3rem;
    --font-bold: 800;

    --font-huge-dark: {
      color: --font-dark;
      font-size: --font-huge;
      font-weight: --font-bold;
    }
  }
```
OR (tachions style abbreviations)
```css
  root: {
    --fc-dark: #4a4a4a;
    --fs-huge: 3rem;
    --fw-bold: 800;

    --font-huge-dark: {
      color: --fc-dark;
      font-size: --fw-huge;
      font-weight: --fw-bold;
    }
  }
```

some-component.css
```css
  .title_bar {
    @apply --font-huge-dark;
  }
```

Kadee80, 2018-01-09:
WIP, first incremental save. Will update with requirements and discussion soon

mcous, 2018-01-12:
Consensus is this RFC is great and we're doing it

🍷

IanLondon, 2018-01-12:
:cocktail:

commit 4610801d (2018-01-13): build(css): Add postcss with cssnext to front-end projects

Closes #591
commit dd79dbdf (2018-01-13): build(css): Add postcss with cssnext to front-end projects

Closes #591
commit 366dbe67 (2018-01-15): build(css): Add postcss with cssnext to front-end projects

Closes #591
commit 70693ca7 (2018-01-15): build(css): Add postcss with cssnext to front-end projects

Closes #591
closed 2018-01-15 by mcous