Opentrons/opentrons #611: Robot doesn't path correctly over 50 mL tube rack

antonjs, 2018-01-16:
## overview

When the robot is configured with a falcon tube container (tube-rack-15_50ml), it doesn't raise the gantry high enough to clear a falcon tube when the pipette has a tip on it, or high enough to enter the top of such a tube.

## behavior

See above. The pipette should raise higher to make it into a falcon tube.

## steps to reproduce

* Use the laser-cut acrylic tube rack (tube-rack-15_50ml).
* Load one or more falcon tubes (for instance, one into A1 and one into A3).
* Calibrate the A1 position to the approximate bottom of the falcon tube.
* Run a protocol that gets a tip and aspirates from the A1 tube.

The robot will pick up a tip and crash it into the side of the A3 tube, followed by not successfully entering the A1 tube.

Using the opentrons 2.5.2 release, running through the app.

## demo protocol
`from opentrons import robot, containers, instruments

p10_tipracks = [containers.load('tiprack-10ul', 'E2')]
p200_tipracks = [containers.load('tiprack-200ul', 'A3')]
falcon = containers.load('tube-rack-15_50ml', 'A1')
trash = containers.load('point', 'D1', 'holywastedplasticbatman')

p10 = instruments.Pipette(
    axis='a',
    max_volume=10,
    min_volume=0.5,
    tip_racks=p10_tipracks,
    trash_container=trash,
    channels=8,
    name='p10-8'
)

p200 = instruments.Pipette(
    axis='b',
    max_volume=200,
    min_volume=20,
    tip_racks=p200_tipracks,
    trash_container=trash,
    channels=1,
    name='p200-1'
)

# robot.arc_height=20
p200.pick_up_tip()
p200.aspirate(100, falcon['A1'])

for c in robot.commands():
    print(c)
`

antonjs, 2018-01-16:
See https://gist.github.com/antonjs/a680bc8e2159c8be4794b54feeb7ef1b

mcous, 2018-07-12:
Likely related to #1849. @btmorr do we need both these tickets?

mcous, 2018-09-23:
No longer observing collision issues with tuberacks (specifically referring to the Opentrons Tube Rack Set, see #2167). There is an outstanding involving with the 15/50ml tuberack where the robot thinks the top of the 50ml tubes is at the same Z-height as the taller 15ml tubes, but it does not result in a collision.

It looks like this issue was filed for the OT1, so please keep in mind that we're not porting these fixes back to the `v2.x` release line. If you're curious, the source code for that line lives in the branch `ot1-stable`

closed 2018-09-23 by mcous

samm-igtivf, 2023-10-25:
> No longer observing collision issues with tuberacks (specifically referring to the Opentrons Tube Rack Set, see #2167). There is an outstanding involving with the 15/50ml tuberack where the robot [...]

@mcous do you have a link to the issue you referenced "involving with the 15/50ml tuberack where the robot thinks the top of the 50ml tubes is at the same Z-height as the taller 15ml tubes"? I believe I'm seeing a similar bug.

mcous, 2023-10-25:
@samm-igtivf I'm not longer an Opentrons employee, and 5 years on I'm don't remember what issue I was referring to there 😅

If you think you're experiencing a geometry bug, I would recommend reaching out to Opentrons support or filing a new issue for the best possible visibility. Quite a bit has changed with the geometry system, especially since PAPI v2.14, so a historical ticket may not be that helpful