Opentrons/opentrons #615: Pipette paths too low when moving inside a plate.

antonjs, 2018-01-16:
## overview

When moving liquid between multiple wells on the same plate, the robot doesn't lift high enough, sometimes dragging the tip(s) across the top of wells while moving.

## behavior

The pipette will raise out of the well, but with minimal or no clearance above the top of a well. It will then move to another well on the same plate, hitting the sides of wells with the tip.

## steps to reproduce

* Set up a protocol with a 96 well plate.
* Transfer from one well on the plate to another well, using the arc strategy.

See https://github.com/EndyLab/opentrons/blob/master/protocols/bug-repro/transfer-height.py.

## workaround

To work around the issue, set the arc_height to something higher than the default of 5:
`robot.arc_height = 20`.

antonjs, 2018-01-16:
This may be the same as #403

umbhau, 2018-02-13:
Closing -- I believe this is fixed with #801.

closed 2018-02-13 by umbhau