Opentrons/opentrons #616: Plate component should display unrecognized labware types without throwing

mcous, 2018-01-16:
## overview

Currently, users have mechanisms through the existing protocol API through which they can add custom labware types to their robot / protocols. Attempting to use these protocols, however, blows up the app because the `Deck` component cannot find the type key in `default-containers.json`.

## behavior

When attempting to calibrate labware with a protocol that uses a non-default labware, the app crashes to a white screen with an exception.

## tasks

- [ ] Update container library to import `default-containers.json` from path in `api` project
    - This is not a great inter-project dependency to have, but seeing as `default-containers.json` is going away:
    - This is better than having two files (one in API and one in PD) that could get out of sync
- [ ] Have plate display some easy, sensible default for custom labware
    - We can worry about the "proper" solution once new labware definitions are locked down

IanLondon, 2018-01-18:
+ Fix `data-wellName` as addressed in TODO:

> `TODO Ian 2017-12-13: should be lowercase data-wellname (Warning: React does not recognize the `data-wellName` prop on a DOM element. If you intentionally want it to appear in the DOM as a custom attribute, spell it as lowercase `data-wellname` instead.)`

IanLondon, 2018-01-18:
[image]

PR #635 (2018-01-19): Complib custom container fallback
closes #616 

## review requests

### testing

Following "steps to reproduce" in #616, create a custom container in the App. Make sure it works and looks acceptable (for now).

Test in PD by manually dispatching the following action in Redux-DevTool:

```js
{
  type: 'CREATE_CONTAINER',
  payload: {
    slotName: '8',
    containerType: 'cool-custom-container-type'
  }
}
```

### funky jest relative path workaround

I added `default-containers.json` to `moduleNameMapper` in `protocol-designer/package.json` to fix a Jest path resolution error:

> `../../api/ope [... cut]
closed 2018-01-19 by IanLondon