Opentrons/opentrons #619: Deck slots 9,10,11 shift location by 10mm on the y axis with multi.

Carlos-fernandez, 2018-01-17:
## overview

## behavior

## steps to reproduce
1. Load multi and tip racks on slots 1,9,10,11
2. Calibrate tip racks based on slot one
3. Run protocol that pickups up a tip, does some action and returns tip.
4. Y axis should be shifted to pick up tips by 10mm dont know why. So it may pick up 7 tips instead 8. Other time it shifts more then that. It goes to the wrong location, its not hardware because i tried this on two robot already. 

Hardware:B2B 
Branch:Demo Vid
Resin OS:2.7.8

Carlos-fernandez, 2018-01-17:
"""
Everything Test script

Author: Carlos Fernandez
"""
#Python 3.5
#Imported Libraries
import os, sys
import time
import datetime
import optparse
import serial
import csv
import itertools

from opentrons import containers, instruments, robot
from opentrons.drivers.smoothie_drivers import driver_3_0

def set_relative():
    robot._driver._send_command('G91')

def set_absolute():
    robot._driver._send_command('G90')
    
def disable_motors():
    robot._driver._send_command('M18')

if __name__ == '__main__':
    #options to pick from
    parser = optparse.OptionParser(usage='usage: %prog [options] ')
    parser.add_option("-c", "--cycles", dest = "cycles", type = 'int', default = 40, help = "Number of Cycles to run")
    parser.add_option("-s", "--speed", dest = "speed", type = 'int', default = 600, help = "Velocity Number mm/s")
    (options, args) = parser.parse_args(args = None, values = None)
    # #File name + directory to record
    file_name = "results/Everything_Test_%s.csv" %(datetime.datetime.now().strftime("%m-%d-%y_%H-%M"))
    print(file_name)
    #robot = driver_3_0.SmoothieDriver_3_0_0()
    #Open CSV file with corresponding headers
    with open(file_name, 'w', newline='') as f:
        test_data = {'Cycles':None,'time':None,'tip_rack':None ,'Errors':None, }
        log_file = csv.DictWriter(f, test_data)
        log_file.writeheader()
        try:
            robot.connect()
            robot.home()
            #Set Max speeds for axes 
            robot._driver._send_command('M203.1 X500 Y500 Z140 A140 B50 C50')  # max axis speed
            #Loads Trough Container on slot B3
            trough_slot = 'B3'
            trough = containers.load('trough-12row', trough_slot)
            #Create a list for 12 dividers
            trough_well = []
            for well in range(12):
                trough_well.append(well)
            print("Trough well: ", trough_well)
            print("trought slot", trough)
            #Load 96 PCR Plate to B2 Slot
            plate_slot = 'B2'
            plate_94 = containers.load('96-PCR-flat', plate_slot)
            plate_well = []
            for well in range(12):
                plate_well.append(well)
            #List of tip rack slots
            slots = ['A1','A2','A3','A4','B1','B4','C1','C2','C3']
            #Load tip racks
            racks = []
            for slot in slots:
                racks.append(containers.load('tiprack-200ul', slot))
            print("tiprack slots: ", racks)
            plate_slots = []
            for slots in range(12):
                plate_slots.append(slots)
            #Name Instrumement and Additional Settings 
            pipette = instruments.Pipette(name='p301-multi', mount='left', tip_racks = racks, channels = 8, aspirate_speed = 5, dispense_speed = 40)
            #Drop tip -5mm
            pipette.plunger_positions['drop_tip'] = -5
            pipette._pick_up_current = 0.4
            #volume to aspirate and dispense
            volume = 200
            for cycle in range(options.cycles):
                print("cycle: ", cycle)
                for rack in racks:
                    rack_location = rack
                    print(rack)
                    for well in plate_well:
                        #print("well: ", well)
                        #print("Pick up tip")
                        pipette.pick_up_tip(presses = 1)#Pick up tip
                        robot._driver._send_command('M906 Z400')
                        robot._driver._send_command('G90 Z75')
                        robot._driver._send_command('G28.2 Z')
                        robot._driver._send_command('G90 Z160')
                        pipette.aspirate(volume, trough['A6'])#aspirate 100
                        pipette.dispense(volume, plate_94.rows(well))
                        #print(plate_94.rows(well))
                        pipette.blow_out()
                        pipette.aspirate(volume, plate_94.rows(well))
                        pipette.dispense(volume, trough['A6'])
                        pipette.blow_out()
                        pipette.return_tip()
                    test_data['Cycles'] = cycle
                    test_data['tip_rack'] = rack
                    test_data['time'] = time.strftime("%H:%M:%S", time.localtime())
                    test_data['Errors'] = None
                    log_file.writerow(test_data)
                    print(test_data)
                    f.flush()
                pipette.reset_tip_tracking()
                
        except KeyboardInterrupt:
            print("Test Cancelled")
            f.flush()
            disable_motors()
        except Exception as e:
            print("ERROR OCCURED")
            test_data['Errors'] = e
            log_file.writerow(test_data)
            f.flush()
            disable_motors()
            raise e

        print("Test done")
        f.flush()
        disable_motor()

btmorr, 2018-01-17:
For clarification, does it always successfully pick up tips, just sometimes a row off from where it should, or does it miss sometimes and try to pick up from some random spot and get no tips at all?

Carlos-fernandez, 2018-01-18:
Slot 9 is a row off, slot 10 and 11 look shift on more toward the front of
the unit by even more then 9mm, which is weird. It does t pick any tips on
10 and 11. On slot 9 it always picks up the tips only on those 7 channels.

andySigler, 2018-01-28:
unable to reproduce on latest `v3a`, running the blow script lines up with all tip racks:
```python
from opentrons import instruments, containers

m50 = instruments.Pipette(
    mount='left',
    channels=8,
    tip_racks=[containers.load('tiprack-200ul', slot) for slot in range(1, 11)]

for i in range(12 * 11):  # num rows per containers, times number of slots
    m50.move_to(m50.get_next_tip())
```

closed 2018-02-05 by andySigler