Opentrons/opentrons #621: Tiprack should be calibrated by the instrument associated with it

btmorr, 2018-01-17:
## overview

When doing calibration, each tiprack should be calibrated using the instrument that is associated with it in the protocol.

## behavior

**Current:** calibration requires a single-channel, and uses it to calibrate everything

**Desired:** for each tiprack, calibration should figure out which instrument is associated with that tiprack, and use that instrument to calibrate it

## steps to reproduce

Create a protocol with two pipettes and at least one tiprack associated with each, and then go through calibration flow

btmorr, 2018-01-17:
Related to #553

btmorr, 2018-01-17:
blocked by #627

mcous, 2018-01-24:
Resized to a large after seeing the amount of work that went into the simpler TipProbe PRs

PR #684 (2018-01-29): Calibrate tipracks with inst. that uses them in protocol
## overview

First off: I'm sorry. This PR is too big. In hindsight:

- Adding flow typings and the required functionality changes in one PR wasn't necessary
- The functionality changes themselves could've been accomplished in two PRs
- Our request/response actions aren't well suited to flow and will need to be changed

Anyway, this PR closes #621 by using the following procedure to select a "calibrator":

- Tipracks will be calibrated by whichever pipette uses that tiprack
- Non-tipracks will be calibrated by whichever pipette still has [... cut]
closed 2018-01-29 by mcous