Opentrons/opentrons #622: App crashes during calibration if no single-channel pipette is present

btmorr, 2018-01-17:
## overview

When trying to calibrate labware for a protocol with no single-channel pipette, the app crashes

## behavior

**Currently:** Calibration flow will crash if no single-channel pipette is configured in the protocol (this is true even if an instrument object is created for a single-channel pipette, if that pipette is not used for any actions)

**Desired:** Calibration should be able to use whatever instruments are present to calibrate labware. Example valid configurations:

- Single channel on right or left mount, none on opposite (currently works)
- Multi channel on right or left mount, none on opposite (currently broken)
- Multi channel on right or left mount, single channel on opposite (currently works)
- Single channel on both mounts (currently works)
- Multi channel on both mounts (currently broken)

## steps to reproduce

Create a protocol with no single-channel pipette actions, and attempt to calibrate labware

btmorr, 2018-01-17:
Related to #553

btmorr, 2018-01-17:
May have actually been taken care of by #572? @mcous can you confirm?

mcous, 2018-01-17:
Yup, it was fixed in #572

closed 2018-01-17 by mcous