Opentrons/opentrons #623: App TipProbe Panels should Match Designs

Kadee80, 2018-01-17:
## overview

Current Designs now call for a single tip probe panel separate from the pipette info area which was addressed in #595. The next step is to update the `TipProbe` container and components to match the design screens and implement cancel continue modal.

## Currently in App:
[image]

## Updated Design:
- `TipProbe` panels can receive a 1 or 2 column layout based on presence of `onClick` prop.  (Button in left column, or entered Spinner)
[image]
[image]
- Cancel/Continue modal displays on first tip probe, or on return to tip probe from labware setup
[image]

## steps to implement

- [ ] Display updated panels based on robot constants:   UNPROBED, PREPARING_TO_PROBE, READY_TO_PROBE, PROBING, PROBED
- Utilize comp lib css exports for all new color/border/typography vars while removing @values
- Implement Cancel/Continue modal on `deckPopulated=true` state atom. #594 partially blocks this

Kadee80, 2018-01-23:
This issue is in PR now. There is a bit of state work needed to get the probe panels to match designs for confirm tip removed, and display a [next] screen before setting probed to true.

mcous, 2018-01-23:
Closed by #652

closed 2018-01-23 by mcous